libimobiledevice/libimobiledevice #1307: ideviceinfo No device found error

anjul0296, 2022-04-26:
Hi,

When i run the ideviceinfo from terminal in arch linux it's give the "Error: No device found" but iPhone already connected in system.

Thanks

wedgef5, 2022-05-15:
I have the same issue on Fedora 36. lsusb shows the device (mine is in Recovery Mode), I have tried restarting usbmuxd multiple times, but nothing seems to work.

Arr0way, 2022-06-28:
> I have the same issue on Fedora 36. lsusb shows the device (mine is in Recovery Mode), I have tried restarting usbmuxd multiple times, but nothing seems to work.

dnf install usbmuxd

Then test with ideviceinfo

Arr0way, 2022-06-28:
> > I have the same issue on Fedora 36. lsusb shows the device (mine is in Recovery Mode), I have tried restarting usbmuxd multiple times, but nothing seems to work.
> 
> dnf install usbmuxd
> 
> Then test with ideviceinfo
> 
> ( @wedgef5 I have just done this with Fedora 36)

digital-slave, 2023-10-25:
system upgrade @AiXanadu