libimobiledevice/libimobiledevice #1310: error: implicit declaration of function 'socket_get_socket_port

sakebook, 2022-05-04:
I triying to build on M1 Mac.

```
$ ./autogen.sh --enable-debug
$ make
```

I face this error.

```
idevicedebugserverproxy.c:278:11: error: implicit declaration of function 'socket_get_socket_port' is invalid in C99 [-Werror,-Wimplicit-function-declaration]
                if (0 > socket_get_socket_port(server_fd, &port)) {
                        ^
1 error generated.
make[2]: *** [idevicedebugserverproxy-idevicedebugserverproxy.o] Error 1
make[1]: *** [all-recursive] Error 1
make: *** [all] Error 2
```

This error occurs after [this commit](https://github.com/libimobiledevice/libimobiledevice/commit/b83bc867e61333b61d2f3366e6daa3f6f988074f).

Because I try same command on [the commit](https://github.com/libimobiledevice/libimobiledevice/commit/8e08d874e157e44bb94c99608fb1c5853631b6b4) , make command is success.

mexmer, 2022-05-04:
@sakebook update also libimobiledevice-glue, function is located there and was added in https://github.com/libimobiledevice/libimobiledevice-glue/commit/5950ea6d4139394c38c7bdc445725d04113c1fee

sakebook, 2022-05-04:
@mexmer 

Thank you.

I update libimobiledevice-glue and make & make install.

After, make command success on libimobiledevice 👍