libimobiledevice/libimobiledevice #1312: How to compile with '-fPIC' in Linux?

HackHarry, 2022-05-07:
My OS version is Ubuntu 18.04 LTS. 

I try to complie my tools depends on libimobiledevice.a, but it failed with this message.
```
/usr/bin/ld: /usr/local/lib/libimobiledevice-1.0.a(idevice.o): relocation R_X86_64_PC32 against symbol 'idevice_connection_disable_bypass_ssl' can not be used when making a shared object; recomplie with -fPIC
/usr/bin/ld: final link failed: Bad value
```
My tools must be complied with '-fPIC', but I don't know how to complie libimobiledevice with -fPIC. I try to add -fPIC to CFLAGS in MakeFile after running './autogen.sh', but it doesn't work.

Besides, linking .so is work for me, but I want to link .a

nikias, 2022-05-11:
It is actually recommended to use shared libraries. If you need to compile with additional flags, you should set them before running `./autogen.sh` so basically if you run `CFLAGS=-fPIC ./autogen.sh` it should add that to the generated Makefiles.

HackHarry, 2022-05-13:
Thanks! It works for me.

closed 2022-05-13 by HackHarry