libimobiledevice/libimobiledevice #1313: No able to run the Apps using command idevicedebug -d run "UDID"

rahul-1485, 2022-05-10:
When i try to run command **idevicedebug -d run com.pcloudy.shake**  in my device i get below error. It happens with all UDID.

apple@Apples-MBP-2 ~ % idevicedebug -d run com.pcloudy.shake
working_directory: /private/var/mobile/Containers/Data/Application/1920EBE1-D9A7-4A6B-ADAE-A3DDE7212470

Could not start com.apple.debugserver!
Please make sure to mount the developer disk image first:
  1) Get the iOS version from `ideviceinfo -k ProductVersion`.
  2) Find the matching iPhoneOS DeveloperDiskImage.dmg files.
  3) Run `ideviceimagemounter` with the above path.

I run the above steps image is mounted but issue is same.

IOS version i am using is 15.0

idevicedebug version 1.3.0

[image]

mexmer, 2022-05-11:
run 
```
idevicemounter -l
```
to see if developer image is really mounted, also keep in mind, you need to mount proper version of developer image. like developer image for iOS 14 will not work with iOS15 and vice versa.
also make sure you use latest version of libimobile (complied from github source), if you working with ios14 or ios15 devices.

rahul-1485, 2022-05-11:
Hi @mexmer  i have installed Xcode 13.2.1 and install ideviceinstaller 1.1.1 but when i run idevicemounter -l it shows below error 

zsh: command not found: idevicemounter

[image]

Can you please guide.

nikias, 2022-05-11:
@rahul-1485 that is a typo, it's called `ideviceimagemounter`. The version 1.3.0 you seem to have is certainly not the latest code, so you would have to build it yourself. Since you are on macOS, this is super easy to do that:
```
mkdir -p limd-build;
cd limd-build;
curl -o ./limd-build-macos.sh -L https://is.gd/3TMOiY ;
bash ./limd-build-macos.sh
```

rahul-1485, 2022-05-11:
thanks @nikias i am able to launch the App now.

closed 2022-05-11 by rahul-1485