libimobiledevice/libimobiledevice #1317: idevicedebugserverproxy - Immediate debugserver connection closed after process connect

TheSacrificer, 2022-05-14:
Hi, I just built the latest version of libplist, usbmuxd, libimobiledevice, ideviceinstaller, libusbmuxd, libimobiledevice-glue and tried to do some remote debugging.

`idevicedebugserverproxy -d 9999`

and tried to connect in lldb by:

`process connect connect://localhost:9999`

but I received immediate debugserver connection closed error:

`% idevicedebugserverproxy -d 9999
main: Handling new client connection...
main: Waiting for connection on local port 9999
connection_handler: client_fd = 4
debugserver connection closed
connection_handler: shutting down...`

Any suggestions???

iOS: 15.4.1 , Xcode: 13.3.1 (13E500a), macOS: 12.3.1 (21E258) idevicedebugserverproxy: 1.3.0-155-gdec0438 ,
lldb-1316.0.9.41: Apple Swift version 5.6 (swiftlang-5.6.0.323.62 clang-1316.0.20.8)

mexmer, 2022-05-16:
do you have developer image mounted inside device?

TheSacrificer, 2022-05-16:
@mexmer yes, and i solved this by uninstalling all libplist, usbmuxd, libimobiledevice, ideviceinstaller, libusbmuxd, libimobiledevice-glue which I built from cloning the latest source, then tried to reinstall them with brew --HEAD with @topeerz 's suggestion in https://github.com/libimobiledevice/libimobiledevice/issues/1217 for  "No package 'libimobiledevice-glue-1.0' found" issue.  

But now there's a new problem, when trying to:
`% idevicedebug  run  com.myapp.mytestapp `
I receive :
`ERROR: failed to get the task for process 1645`
Then the app starts but not fully loaded and stuck at very first screen.
Any suggestion??

TheSacrificer, 2022-05-16:
sorry, misclicked "Close with comment"

TheSacrificer, 2022-05-16:
the solution launching 4 times in a row in https://github.com/libimobiledevice/libimobiledevice/issues/535 does'nt work for me

mexmer, 2022-05-16:
most likely proces is starting too slow, and debugger times out, before being able to attach to process

TheSacrificer, 2022-05-16:
@mexmer Thanks for reply! Then is there a way to set a longer time out?

HMaker, 2023-06-10:
@TheSacrificer you fixed it? I have the exactly same issue with `idevicedebug`
```
❯ /usr/local/bin/idevicedebug --debug run com.example.TheApp
working_directory: /private/var/mobile/Containers/Data/Application/35E49981-0B3F-4843-983E-CB3DBF37B10E
Setting logging bitmask...
Setting maximum packet size...
Setting working directory...
Setting argv...
app_argv[0] = /private/var/containers/Bundle/Application/35E49981-0B3F-4843-983E-CB3DBF37B10E/TheApp.app/TheApp
Checking if launch succeeded...
ERROR: failed to get the task for process 428
```