libimobiledevice/libimobiledevice #1321: cycle in include mapping with trivial one line fix.

kkuehl, 2022-05-30:
Using the [https://github.com/include-what-you-use/include-what-you-use](https://github.com/include-what-you-use/include-what-you-use) utility against one of my projects that utilizes libimobiledevice, I came accross this issue.

```Error running 'include-what-you-use': Cycle in include-mapping:
  <libimobiledevice/lockdown.h> ->
  <libimobiledevice/lockdown.h>
```
This was trivially resolved with the following change:
```
diff --git a/include/libimobiledevice/lockdown.h b/include/libimobiledevice/lockdown.h
index f22917a..6d390fd 100644
--- a/include/libimobiledevice/lockdown.h
+++ b/include/libimobiledevice/lockdown.h
@@ -31,7 +31,6 @@ extern "C" {
 #endif
 
 #include <libimobiledevice/libimobiledevice.h>
-#include <libimobiledevice/lockdown.h>
 
 /** Error Codes */
 typedef enum {
dd
```

mexmer, 2022-05-31:
make PR for that ... seems like copypasta typo

luckily for preprocessor, include guards will not allow recursive processing, so it's used only once ... but should not be there indeed.

nikias, 2022-05-31:
No need for PR. I will just fix it.

nikias, 2022-05-31:
https://github.com/libimobiledevice/libimobiledevice/commit/5b4ba89b1f86c9194a43a520b9b49faebffd784a

closed 2022-05-31 by nikias