libimobiledevice/libimobiledevice #1322: Merger lib with Xcode Swift Project - dylib was built for newer macOS version (10.15) than being linked (10.13)

danio0701, 2022-06-06:
Hi guys, I fight over 80h with adding libimobiledevice libs to xcode project. I succeed with that and all works on catalina and up but on older systems like high sierra or mojave app crash with errors

`dyld: Symbol not found: ___darwin_check_fd_set_overflow
  Referenced from: /Applications/XXXXXXXX.app/Contents/MacOS/../Frameworks/libimobiledevice-glue-1.0.0.dylib (which was built for Mac OS X 11.0)
  Expected in: /usr/lib/libSystem.B.dylib
 in /Applications/AMPRO Hello.app/Contents/MacOS/../Frameworks/libimobiledevice-glue-1.0.0.dylib
Abort trap: 6
`

same for most of libs I have (errors for them also):

`
Showing All Messages
Dylib (PATH_XXXXX/libimobiledevice/libcrypto.1.1.dylib) was built for newer macOS version (10.15) than being linked (10.13)
Dylib (PATH_XXXXX/libimobiledevice/libcrypto.3.dylib) was built for newer macOS version (10.15) than being linked (10.13)
Dylib (PATH_XXXXX/libimobiledevice/libideviceactivation-1.0.2.dylib) was built for newer macOS version (11.0) than being linked (10.13)
Dylib (PATH_XXXXX/libimobiledevice/libideviceactivation-1.0.dylib) was built for newer macOS version (11.0) than being linked (10.13)
Dylib (PATH_XXXXX/libimobiledevice/libimobiledevice-1.0.6.dylib) was built for newer macOS version (11.0) than being linked (10.13)
Dylib (PATH_XXXXX/libimobiledevice/libimobiledevice-1.0.dylib) was built for newer macOS version (11.0) than being linked (10.13)
Dylib (PATH_XXXXX/libimobiledevice/libimobiledevice-glue-1.0.0.dylib) was built for newer macOS version (11.0) than being linked (10.13)
Dylib (PATH_XXXXX/libimobiledevice/libimobiledevice-glue-1.0.dylib) was built for newer macOS version (11.0) than being linked (10.13)
Dylib (PATH_XXXXX/libimobiledevice/libirecovery-1.0.3.dylib) was built for newer macOS version (11.0) than being linked (10.13)
Dylib (PATH_XXXXX/libimobiledevice/libirecovery-1.0.dylib) was built for newer macOS version (11.0) than being linked (10.13)
Dylib (PATH_XXXXX/libimobiledevice/libplist-2.0.3.dylib) was built for newer macOS version (10.15) than being linked (10.13)
Dylib (PATH_XXXXX/libimobiledevice/libplist-2.0.dylib) was built for newer macOS version (10.15) than being linked (10.13)
Dylib (PATH_XXXXX/libimobiledevice/libplist++-2.0 2.dylib) was built for newer macOS version (10.15) than being linked (10.13)
Dylib (PATH_XXXXX/libimobiledevice/libplist++-2.0.2.dylib) was built for newer macOS version (10.15) than being linked (10.13)
Dylib (PATH_XXXXX/libimobiledevice/libplist++-2.0.3.dylib) was built for newer macOS version (10.15) than being linked (10.13)
Dylib (PATH_XXXXX/libimobiledevice/libplist++-2.0.dylib) was built for newer macOS version (10.15) than being linked (10.13)
Dylib (PATH_XXXXX/libimobiledevice/libssl.1.1.dylib) was built for newer macOS version (10.15) than being linked (10.13)
Dylib (PATH_XXXXX/libimobiledevice/libssl.3.dylib) was built for newer macOS version (10.15) than being linked (10.13)
Dylib (PATH_XXXXX/libimobiledevice/libusbmuxd-2.0.6.dylib) was built for newer macOS version (10.15) than being linked (10.13)
Dylib (PATH_XXXXX/libimobiledevice/libusbmuxd.4.dylib) was built for newer macOS version (10.14) than being linked (10.13)`

How to make it working on 10.13 and newer systems? Where can get fully working libs?

mexmer, 2022-06-07:
build them from source code, then you can run it.

danio0701, 2022-06-07:
I was trying already but on my mac always multiple of errors. I got also binary file from other place but it works only on my catalina but on older systems missing libs always. I need to make binary which have all libs included or put all to app project

> build them from source code, then you can run it.

mexmer, 2022-06-07:
@nikias  wrote skript for building libimobile on os X, i'm not sure on which version it will work, but since it also downloads lot of required stuff, might work for you

then you can use resulting libraries in your xcode projects

otherwise i would suggest to contact author of binaries you used, to send you versions for older systems.

```sh
mkdir -p limd-build
cd limd-build
curl -o ./limd-build-macos.sh -L is.gd/kdT2Nl
bash ./limd-build-macos.sh
```