libimobiledevice/libimobiledevice #1332: Can't restore iphone 7 backup to new iphone

pompass, 2022-06-21:
I was able to successfully backup my old iphone 7:  

idevicebackup2 backup --full /home/dan/Documents/7

and tried restoring backup to new iphone with the command below: 

idevicebackup2 -i -s UUID restore --system /home/dan/Documents/7

However, when entering the restore command I get output related to the different commands and flags available under idevicebackup2. Both iphones have software version 15.5
what am I doing wrong?

Thanks

nikias, 2022-06-21:
Hmm can't spot anything right away. Check the first line of the output, it usually has an error description saying what's wrong.

pompass, 2022-06-21:
This is the output (phone numeric UDID was entered):

dan@pc:~$ idevicebackup2 -i -s UDID restore —system /home/dan/Documents/7s
Usage: idevicebackup2 [OPTIONS] CMD [CMDOPTIONS] DIRECTORY

Create or restore backup from the current or specified directory.

CMD:
  backup create backup for the device
    --full force full backup from device.
  restore restore last backup to the device
    --system restore system files, too.
    --no-reboot do NOT reboot the device when done (default: yes).
    --copy create a copy of backup folder before restoring.
    --settings restore device settings from the backup.
    --remove remove items which are not being restored
    --skip-apps do not trigger re-installation of apps after restore
    --password PWD supply the password of the source backup
  info show details about last completed backup of device
  list list files of last completed backup in CSV format
  unback unpack a completed backup in DIRECTORY/_unback_/
  encryption on|off [PWD] enable or disable backup encryption
    NOTE: password will be requested in interactive mode if omitted
  changepw [OLD NEW]  change backup password on target device
    NOTE: passwords will be requested in interactive mode if omitted
  cloud on|off enable or disable cloud use (requires iCloud account)

OPTIONS:
  -u, --udid UDID target specific device by UDID
  -s, --source UDID use backup data from device specified by UDID
  -n, --network connect to network device
  -i, --interactive request passwords interactively
  -d, --debug enable communication debugging
  -h, --help prints usage information
  -v, --version prints version information

Homepage:    <https://libimobiledevice.org>
Bug Reports: <https://github.com/libimobiledevice/libimobiledevice/issues>
dan@pc:~$

nikias, 2022-06-22:
That's not the latest code, I can tell from the output, but I still can't spot a mistake. Also, I would suggest to also pass `--settings` option additionally to `--system` whenever you get it to work so you keep your configured settings.

mexmer, 2022-06-22:
usage print without error means unrecongnized input parameter, i expect some typo in UDID (some other character than hexadeciemal numbers) that causes it to treat it as multiple entities instead of single parameter value.

pompass, 2022-06-22:
Thank you nikias and mexmer, I'm pretty sure the UDID was correct but will restore again and paste the output/error here for review.

pompass, 2022-06-22:
restore not working, both phones were paired before backup and restore.

I did a full and successful backup (18GB) of the old phone without specifying it's UDID with the command below: 

	idevicebackup2 backup --full /home/dan/Documents/7

then tried to do a restore into the new phone using the old phone's UDID with no success: 

	idevicebackup2 -i -s 123456789abcdef12345678 restore —system /home/dan/Documents/7

any suggestions, should I enter the commands differently?

mexmer, 2022-06-23:
just tested on linux  (debian) and it worked as expected ... of course i did use latest version compiled from source, because i have 15.5 on phone (and yes, it's iphone 7)

pompass, 2022-06-23:
I have the same setup, debian 11.3.0 and both iphones 7 and 12 have v 15.5.
but, I installed libimobiledevice and usbmuxd through synaptic and didn't compiled from source.

nikias, 2022-06-23:
Are you passing `-system` or `--system` ? because it needs to have two dashes: `--system`. Also don't forget `--settings`.

pompass, 2022-06-23:
yes, passing --system with two dashes.
I didn't pass --settings, where should it fit in the command below?

     idevicebackup2 -i -s 123456789abcdef12345678 restore --system /home/dan/Documents/7

nikias, 2022-06-23:
```idevicebackup2 -i -s 123456789abcdef12345678 restore --system --settings /home/dan/Documents/7```