libimobiledevice/libimobiledevice #1333: Unable to debug in iOS 16 developer beta 2

JJTech0130, 2022-06-22:
I get "Could not start com.apple.debugserver!
Please make sure to mount the developer disk image first:
  1) Get the iOS version from `ideviceinfo -k ProductVersion`.
  2) Find the matching iPhoneOS DeveloperDiskImage.dmg files.
  3) Run `ideviceimagemounter` with the above path." even though it is clearly mounted.

BluDood, 2022-06-22:
Same issue here. Seems like Apple did changes to debugserver

jkcoxson, 2022-06-23:
Many users of JitStreamer are also reporting this, which is based on libimobiledevice.

JJTech0130, 2022-06-23:
And here it is from AltJIT (note that the log may contain irrelevant data:
```
error	10:44:11.332501-0400	backboardd	ImageQueueCollectable client message err=0x10000004
default	10:44:11.332623-0400	cloudd	BEGIN [7190]: FetchAnisette
error	10:44:11.334806-0400	backboardd	ImageQueueCollectable client message err=0x10000004
error	10:44:11.337408-0400	backboardd	ImageQueueCollectable client message err=0x10000004
error	10:44:11.340092-0400	backboardd	ImageQueueCollectable client message err=0x10000004
error	10:44:11.342788-0400	backboardd	ImageQueueCollectable client message err=0x10000004
error	10:44:11.345366-0400	backboardd	ImageQueueCollectable client message err=0x10000004
default	10:44:11.345519-0400	cloudd	END [7190] 0.013234s:FetchAnisette
error	10:44:11.347794-0400	backboardd	ImageQueueCollectable client message err=0x10000004
default	10:44:11.348889-0400	cloudd	Task <37E95C76-DBD3-4844-9E1A-D96C9B2042DA>.<6> authenticator finished cont 1 req (null) headers <private>
default	10:44:11.350323-0400	cloudd	BackgroundSession <4FAD8826-2890-43A9-AEEB-8A97D53ED57F> BackgroundDataTask <37E95C76-DBD3-4844-9E1A-D96C9B2042DA>.<6> finished getting auth headers for session <private> with identifier <private>
error	10:44:11.350488-0400	backboardd	ImageQueueCollectable client message err=0x10000004
default	10:44:11.351406-0400	nsurlsessiond	NDSession <4FAD8826-2890-43A9-AEEB-8A97D53ED57F> Task <37E95C76-DBD3-4844-9E1A-D96C9B2042DA>.<6> received auth headers for response, continueLoad: 1 authHeaders <private> newRequest (null)
default	10:44:11.351691-0400	nsurlsessiond	Task <37E95C76-DBD3-4844-9E1A-D96C9B2042DA>.<6> authenticator finished cont 1 req (null) headers <private>
default	10:44:11.352005-0400	nsurlsessiond	Task <37E95C76-DBD3-4844-9E1A-D96C9B2042DA>.<6> protocol received continueLoad=1, headers=<private> req=(null)
default	10:44:11.352606-0400	nsurlsessiond	Task <37E95C76-DBD3-4844-9E1A-D96C9B2042DA>.<6> {strength 0, tls 4, sub 0, sig 1, ciphers 0, bundle 1, builtin 0}
error	10:44:11.352885-0400	backboardd	ImageQueueCollectable client message err=0x10000004
default	10:44:11.353916-0400	nsurlsessiond	Task <37E95C76-DBD3-4844-9E1A-D96C9B2042DA>.<6> TAKING power assertion.
default	10:44:11.354276-0400	nsurlsessiond	Connection 339: working for activity <private>
default	10:44:11.354577-0400	nsurlsessiond	Connection 339: working for activity <private>
default	10:44:11.355298-0400	nsurlsessiond	Task <37E95C76-DBD3-4844-9E1A-D96C9B2042DA>.<6> now using Connection 339
error	10:44:11.355422-0400	backboardd	ImageQueueCollectable client message err=0x10000004
default	10:44:11.357609-0400	nsurlsessiond	[C339] event: client:start_activity @0.540s, activity <nw_activity 0:0[95F4CE27-862F-4539-8ECC-8E7D243324BD] (lightweight) (reporting strategy default)>
error	10:44:11.358067-0400	backboardd	ImageQueueCollectable client message err=0x10000004
default	10:44:11.360699-0400	AltStore	Task <1E3B27DD-E54B-4105-927F-60EE495FAAE3>.<3> resuming, timeouts(60.0, 604800.0) QOS(0x19) Voucher (null)
default	10:44:11.361093-0400	nsurlsessiond	Task <37E95C76-DBD3-4844-9E1A-D96C9B2042DA>.<6> sent request, body S 653
default	10:44:11.361238-0400	AltStore	[Telemetry]: Activity <nw_activity 12:2[13B7DC66-AE8A-4758-8C6A-EB14E793A971] (reporting strategy default)> on Task <1E3B27DD-E54B-4105-927F-60EE495FAAE3>.<3> was not selected for reporting
default	10:44:11.361402-0400	AltStore	Task <1E3B27DD-E54B-4105-927F-60EE495FAAE3>.<3> {strength 1, tls 8, sub 0, sig 0, ciphers 1, bundle 0, builtin 0}
default	10:44:11.361499-0400	AltStore	Task <1E3B27DD-E54B-4105-927F-60EE495FAAE3>.<3> now using Connection 8
error	10:44:11.361717-0400	backboardd	ImageQueueCollectable client message err=0x10000004
default	10:44:11.361774-0400	AltStore	Task <1E3B27DD-E54B-4105-927F-60EE495FAAE3>.<3> sent request, body N 0
error	10:44:11.363137-0400	backboardd	ImageQueueCollectable client message err=0x10000004
error	10:44:11.365695-0400	backboardd	ImageQueueCollectable client message err=0x10000004
error	10:44:11.368228-0400	backboardd	ImageQueueCollectable client message err=0x10000004
default	10:44:11.369230-0400	symptomsd	Data Usage for lockdownd on flow 48321 - WiFi in/out: 254107794/28486482430, WiFi delta_in/delta_out: 3282/3096, Cell in/out: 0/0, Cell delta_in/delta_out: 0/0, RNF: 0, subscriber tag: 0
error	10:44:11.370745-0400	backboardd	ImageQueueCollectable client message err=0x10000004
error	10:44:11.373277-0400	backboardd	ImageQueueCollectable client message err=0x10000004
default	10:44:11.375204-0400	analyticsd	[TransformManager::checkpoint] 44 transforms persisted. 0 failures.
error	10:44:11.375818-0400	backboardd	ImageQueueCollectable client message err=0x10000004
error	10:44:11.378350-0400	backboardd	ImageQueueCollectable client message err=0x10000004
error	10:44:11.380891-0400	backboardd	ImageQueueCollectable client message err=0x10000004
error	10:44:11.383419-0400	backboardd	ImageQueueCollectable client message err=0x10000004
error	10:44:11.386101-0400	backboardd	ImageQueueCollectable client message err=0x10000004
default	10:44:11.386684-0400	analyticsd	[Daemon perf] Performance checks are disabled. Please update the daemon perf params in defaults if you wish to turn this on.
error	10:44:11.388497-0400	backboardd	ImageQueueCollectable client message err=0x10000004
error	10:44:11.391033-0400	backboardd	ImageQueueCollectable client message err=0x10000004
error	10:44:11.393579-0400	backboardd	ImageQueueCollectable client message err=0x10000004
[... cut ...]  high power state: no, early fire constant interval 0.000000
default	10:44:15.421403-0400	identityservicesd	Invalidating timer <PCPersistentTimer: 0x106291d90>
default	10:44:15.421534-0400	identityservicesd	Invalidating simple timer <PCSimpleTimer: 0x10629ddc0>
default	10:44:15.443581-0400	identityservicesd	Disabling power monitoring for <PCSimpleTimer: 0x10629ddc0> - 3 timers
default	10:44:15.443677-0400	identityservicesd	<PCPersistentTimer: 0x106291d90> created <PCSimpleTimer: 0x10629ddc0>
```

truonggiang0710, 2022-07-12:
@JJTech0130 ,
I think you need to enable developer mode on iOS16. https://developer.apple.com/documentation/xcode/enabling-developer-mode-on-a-device
Hi @nikias  We need to connect the device the first time to Xcode, to show the Developer mode menu. Can libimobiledevice do like that?

JJTech0130, 2022-07-12:
@truonggiang0710 Assuming you mean the new setting in Privacy and Security, then developer mode is enabled. It has to be, because I can install develop singed apps just fine. I can even debug them in XCode, just not libimobiledevice.

kdbeck, 2022-07-16:
ideviceimagemounter has same issue with Developer Mode.

New devmodectl command in macOS 13 enables Developer Mode and reboots connected iPhone without a passcode.
Is it possible to support similar command in libimobiledevice?

JJTech0130, 2022-07-16:
> New devmodectl command in macOS 13 enables Developer Mode and reboots connected iPhone without a passcode. Is it possible to support similar command in libimobiledevice?

You should open a separate issue for that.

jkcoxson, 2022-07-16:
It would be pretty easy though

nikias, 2022-07-17:
https://gist.github.com/nikias/262bd709c1651e0139eb9e3a2e2d33f4

JJTech0130, 2022-07-17:
Um, this issue is kinda turning into a discussion on activating developer mode, which is not the cause of this issue…

jkcoxson, 2022-07-17:
Thanks, that will be super simple to implement. No new services, just some lockdown commands 😅

nikias, 2022-07-17:
I am not sure where to put it, because I don't think we need another tool... but I guess none of the existing ones are a good match...

truonggiang0710, 2022-07-18:
Hi @nikias ,
On my phone,
I ran the command and it did not enable developer mode.
Show the error as below
"
<?xml version="1.0" encoding="UTF-8"?>
<!DOCTYPE plist PUBLIC "-//Apple//DTD PLIST 1.0//EN" "http://www.apple.com/DTDs/PropertyList-1.0.dtd">
<plist version="1.0">
<dict>
	<key>action</key>
	<integer>2</integer>
</dict>
</plist>

<?xml version="1.0" encoding="UTF-8"?>
<!DOCTYPE plist PUBLIC "-//Apple//DTD PLIST 1.0//EN" "http://www.apple.com/DTDs/PropertyList-1.0.dtd">
<plist version="1.0">
<dict>
	<key>Error</key>
	<string>An unknown error has occured</string>
</dict>
</plist>
"

truonggiang0710, 2022-07-20:
Hi @nikias , does the command run successfully on the phone that never has first connect to XCode 14 beta?

nikias, 2022-07-20:
I copied the behavior from the official tool, nothing more, nothing less. I haven't tested it.

jkcoxson, 2022-08-05:
@JJTech0130 I think this can be closed, as this was proved to be false.

JJTech0130, 2022-08-05:
What actually was causing it then? Was it just a bug in B2?

airwolf2026, 2022-09-19:
May be there a new service to mount the  DeveloperDiskImage on iOS16?

nikias, 2022-09-19:
No, you need to enable Developer Mode on the device first, then it mounts just fine.

mexmer, 2022-09-19:
@airwolf2026 no new service, it's new security measure set by apple for ios 16

even Xcode needs developer mode, to be able to debug and sideload apps to ios 16 device
to enable developer mode see
https://developer.apple.com/documentation/xcode/enabling-developer-mode-on-a-device

menu option for enabling developer mode can be enabled by Xcode

rpayne381, 2022-10-03:
Does anyone have a PR to get https://gist.github.com/nikias/262bd709c1651e0139eb9e3a2e2d33f4 idevicedevmodectl working in libimobiledevice?

mexmer, 2022-10-04:
fixed version is here
https://gist.github.com/mexmer/5c9dc1220a9d028d017d1eaa76fe8df2

nikias, 2022-10-04:
I am working on integrating the tool which will then also wait for device reboot etc. but for now use the updated version from @mexmer .

rpayne381, 2022-10-04:
thanks @nikias.

JJTech0130, 2022-10-04:
Has this issue been taken over for discussing the new `devmodectl`? I guess I'll rename it...

mexmer, 2022-10-04:
i think you could have closed this issue.
i mean, your problem was solved by using idevicedevmodectl tool from compiled gist.

nikias is now working on better version of tool, but that's another topic 😹

JJTech0130, 2022-10-04:
Well, actually, my problem *wasn't* solved by it. It seemed like it was just a bug in Developer Beta 2 that caused the issue- it works fine in later betas.
For some reason people brought it up in this issue, I'm not really sure why.

JJTech0130, 2023-01-05:
Completed in https://github.com/libimobiledevice/libimobiledevice/commit/a6775bc588db13838bebec42b139748d337e7189

closed 2023-01-05 by JJTech0130