libimobiledevice/libimobiledevice #1335: ideviceinfo -n dont find device

developer-rdg, 2022-06-29:
[image]

### Can someone help me find out what is wrong with my configuration,
I install **libimobiledevice 1.3.0** in **Raspberry Pi 4** using the command 
`sudo apt-get install usbmuxd libimobiledevice6 libimobiledevice-utils`

I want to manage my iPhone by Wifi but when i type `ideviceinfo -n` always give back the error : _ERROR: No device found!_
Both Raspberry Pi and the **iPhone SE 2 Gen** are in the same wifi network.
When the iPhone is connect by USB and type `ideviceinfo` works fine.

Can someone help me with this problem?
Thank you in advance.

mexmer, 2022-06-29:
network discovery is supported only under OS X. it doesn't work under linux.

there exists alternate usbmuxd with avahi support, that should work ... but had not tested it
https://github.com/tihmstar/usbmuxd2

superrnovae, 2022-07-01:
There's also netmuxd https://github.com/jkcoxson/netmuxd with zeroconf support

Demuos, 2022-07-13:
I do have the same problem running Ubuntu 22.04. I Followed supernovae advice and setup netmuxd, which is running as extension to usbmuxd.
`1|netmuxd- | Starting netmuxd
1|netmuxd- | Starting mDNS discovery
1|netmuxd- | Listening on 127.0.0.1:27015
1|netmuxd- | WARNING: Running in host mode will not work unless you are running a daemon in unix mode as well
1|netmuxd- | Adding device 00008030-xxx`
I also set the environment variable like `export USBMUXD_SOCKET_ADDRESS=127.0.0.1:27015`
but still no device is found `ideviceinfo -n
ERROR: No device found!` Do I'm missing something or how can I debug the connection?

mexmer, 2022-07-14:
this is more question to author of netmuxd

Geczy, 2023-06-17:
will `ideviceinstaller` never be supported on a pi?