libimobiledevice/libimobiledevice #1340: IOS Esim Erasure/Wipe

ClaireMane, 2022-07-13:
How to erase esim using libimobiledevice? 

I have tried the erase option in idevicebackup2, but it did not work. Do I have to send a certain key with the erase? Is there any other tool in libimobiledevice that I can use to erase the esim?

hiepnh0, 2023-03-14:
What command you run?

AndreiCMS, 2024-06-14:
"idevicebackup2 erase" i think, from ethana's build (https://github.com/ethanatwork/libimobiledevice). I am also using this command, but didn't find anything related to automated eSIM erasure.

tudorcraciun, 2024-06-28:
Hello @AndreiCMS, have you managed to find anything in this direction? I believe you have to send the `PreserveDataPlan` parameter as `false` with the `EraseDevice` command, but I haven't managed to test it yet.