libimobiledevice/libimobiledevice #1343: [HELP] lldb wont connect to process

darbysauter, 2022-07-22:
```
$ idevicedebugserverproxy --lldb -d 1234

$ idevicesyslog

$ lldb
(lldb) platform select remote-ios --sysroot /Applications/Xcode-beta.app/Contents/Developer/Platforms/iPhoneOS.platform/Developer/SDKs/iPhoneOS16.0.sdk
(lldb) platform connect connect://localhost:1234
(lldb) attach 1392
```

Eventually times out and prints:
```
Process 1392 stopped
* thread #1, queue = 'com.apple.main-thread', stop reason = signal SIGSTOP
    frame #0: 0x00000001c3b28214 libsystem_kernel.dylib`mach_msg2_trap + 8
libsystem_kernel.dylib`mach_msg2_trap:
->  0x1c3b28214 <+8>: ret

libsystem_kernel.dylib`macx_swapon:
    0x1c3b28218 <+0>: mov    x16, #-0x30
    0x1c3b2821c <+4>: svc    #0x80
    0x1c3b28220 <+8>: ret
Target 0: (RustIos) stopped.

Executable module set to "/private/var/containers/Bundle/Application/8D771D57-93BA-4D05-8065-ADF9C44EA74C/rust_ios.app/RustIos".
Architecture set to: arm64-apple-ios.
```

Output of idevicedebugserverproxy: 
```
main: Waiting for connection on local port 1234
main: Handling new client connection...
main: Waiting for connection on local port 1234
connection_handler: client_fd = 4
main: Handling new client connection...
main: Waiting for connection on local port 1234
connection_handler: client_fd = 6
```

```
(lldb) platform status
  Platform: remote-ios
    Triple: arm64-apple-ios
OS Version: 16.0.0
  Hostname: (null)
 Connected: yes
  SDK Path: "/Applications/Xcode-beta.app/Contents/Developer/Platforms/iPhoneOS.platform/Developer/SDKs/iPhoneOS16.0.sdk"
 SDK Roots: [ 0] "/Applications/Xcode-beta.app/Contents/Developer/Platforms/iPhoneOS.platform/Developer/SDKs/iPhoneOS16.0.sdk"
```

darbysauter, 2022-07-23:
nvm i was thinking it was sig_kill. but its taking 5+ minutes to attach.

moskaliukzhanna, 2024-07-02:
HI @darbysauter  
I am having similar issue. 
I am using xcrun devicectl and then connecting lldb to process it exits with the same error 
`Process 1392 stopped
* thread #1, queue = 'com.apple.main-thread', stop reason = signal SIGSTOP
    frame #0: 0x00000001c3b28214 libsystem_kernel.dylib`mach_msg2_trap + 8
libsystem_kernel.dylib`mach_msg2_trap:
->  0x1c3b28214 <+8>: ret

libsystem_kernel.dylib`macx_swapon:
    0x1c3b28218 <+0>: mov    x16, #-0x30
    0x1c3b2821c <+4>: svc    #0x80
    0x1c3b28220 <+8>: ret
Target 0: (RustIos) stopped.`

What do you mean by saying you are able to continue afterwards ?