libimobiledevice/libimobiledevice #1344: Idevicebackup2 no encryption backup can’t read manifest.db

mark7699, 2022-07-28:
Backed up iPhone using idevicebackup2 with a non encrypted backup. What ever I try everything says manifest.db is not a database? Any clues why? Have tried iLeapp and others and simple SQL lite database reader all come back with same manifest.db is not a database. What am I missing?

mexmer, 2022-07-28:
open manifest.db in simple hexviewer, and se if header at begining of file is sqlite header.

also, what version of iOS you are running on phone?

mexmer, 2022-08-01:
tested on ios 15 and 16 and database is not encrypted.
you can check, if encryption is enabled by running 
```
ideviceinfo -q com.apple.mobile.backup
```
output might look like this
```
CloudBackupEnabled: true
LastiTunesBackupDate: 681028961
LastiTunesBackupTZ: GMT+2
RequiresEncryption: 0
Version: 2.0
WillEncrypt: false
```
WillEncrypt is value of your interest, if it's true, then backup will be encrypted (that also applies to manifest database since ios 10.2).

Gulahmad1997, 2024-07-24:
hi i have an itunes backup file have an issue with it can anyone please help me 

i started backup encrypted with password i have it and the backup is 101GB Size 
the problem is when i restore the backup on any iphone with latest ios also same ios as 17.5.1 the backup sometimes restore fast like its some MB files backup and some times it take few hours to restore bat after restore on ios device i can only see the home screen apps blank need to get downloaded but no other files all pics videos all other files not saving on iphone any help please thamks i also have whatsapp number if someone welling to help me fix this thanks