libimobiledevice/libimobiledevice #1348: Reverse tethering possible?

mfide, 2022-08-12:
Hi,

I have installed this tool into my embedded platform and I could share my phone's internet with it over the USB cable. (by enabling hot spot)

But what I want is that to share my Linux platform's internet with my phone. Similar to one they did with MAC [here](https://apple.stackexchange.com/questions/88449/can-i-share-my-imacs-internet-to-iphone-or-ipad-over-usb)

Is it possible?

JJTech0130, 2023-01-05:
This sounds really useful, especially for situations where you want to share a WiFi connection (such as Hotel or Airport WiFI) that only allows you to connect a single device.

rhysperry111, 2023-04-03:
Just bumping this because it would be extremely useful. I use VNC for my iPad to be a second screen for my monitor, and it would be nice to be able to improve the quality and decrease the latency by having it be wired.

JJTech0130, 2024-05-05:
This is actually already implemented, just not well documented. See https://apple.stackexchange.com/a/455343 and https://github.com/libimobiledevice/usbmuxd/issues/205
Basically, you just need to compile usbmuxd (and the it's dependencies like libimobiledevice) from source, and modify the systemd service to turn that feature on.
Once properly enabled, it should show up as a new `usb0` interface which you can add to a normal bridge to share internet with.

hyphenlee, 2024-09-11:
@JJTech0130  I got this working with iphone 14 and debian 12,
here is the step: [link](https://apple.stackexchange.com/a/475254/552356)

ryan-rosello, 2025-01-09:
@muvaf I get a 100mb/s connection following these instructions, but the label is USB Ethernet, not Apple Ethernet. Did you need to change anything else like ip forwarding?

0xbaksa, 2026-09-04:
I've made a Windows port of this, using in box drivers: https://github.com/0xbaksa/iphone-usb-reverse-tethering-windows

Same 0x52 / mode 3 switch (parameters from usbmuxd's usb.c), sent through a libusb-win32 filter driver.