libimobiledevice/libimobiledevice #1349: No IP assigned without hotspot?

mfide, 2022-08-12:
Hi,

I want to create a local network between my iPad and Embedded Linux Platform over the USB connection. 
I have added all libimobiledevice recipes using yocto, they are all fine.

When I connect my phone to the Linux over the USB cable, linux recognizes it automatically, ipther driver loaded and new ethernet interface eth1 created automatically for it. All good.

But if I don't enable hot spot option on my iPad (which should not be needed just for a local network), my linux platform can't get an answer to its dhcp ip assignment request (udhcpc -i eth1). This works once I enable hot spot on the iPad and IP assigned automatically for it as 172.20.10.7.

But I couldn't make it work without hotspot. The Hotspot option is only available if you have mobile service and not all iPads have that.

I also try to enable dhcp server on my embedded linux platform for eth1 interface so that in case iPad request an IP, it would be assigned automatically but that never happens also.

Is it possible to create a just local network between linux and iPad over the USB cable without enabling the hotspot ?

Thank you.

mexmer, 2022-08-15:
no, it's limitation of ios.
on jailbroken device, you could possibly force it, but on normal device usb tethering without hotspot is not available.

mfide, 2022-08-15:
@mexmer: Actually with iMac it works and iPad gets ip automatically from Mac from 192.168.2.1 range. https://apple.stackexchange.com/questions/88449/can-i-share-my-imacs-internet-to-iphone-or-ipad-over-usb

mexmer, 2022-08-15:
@mfide that is reverse tethering, when macos shares internet with ios device, but afaik it did work only on older osX version, not the new ones (at least noone was able to replicate on bigsur or monterey).

mfide, 2022-08-15:
@mexmer I have been tested it with Macbook Pro M1 13' running latest current MacOS Monterey 12.5and iPad Pro 2020 version, running latest current iOS 15.6. Mac IP is 192.168.2.1 and iPad ip asseigned as 192.168.2.4

mexmer, 2022-08-15:
good to know, that reverse tethering could work for op.