libimobiledevice/libimobiledevice #1355: SHA-1 getting obsoleted in distributions

hadess, 2022-09-07:
Fedora, amongst others, is removing SHA-1 support for signatures (eventually) from gnutls and openssl:
https://fedoraproject.org/wiki/Changes/StrongCryptoSettings3Forewarning2

iOS has removed support for SHA-1 TLS certificates, but it obviously still supports that for communication with iOS devices.

So two questions:
- can the SHA-1 signing be transparently replaced with, say, SHA-256 and still support connecting to an iPhone 3GS
- is this signing relevant when connecting to an iDevice over Wi-Fi, as [usbmuxd2 allows](https://github.com/tihmstar/usbmuxd2)?

Downstream bug:
https://bugzilla.redhat.com/show_bug.cgi?id=2124349

hadess, 2022-09-07:
This change might be needed:
https://github.com/libimobiledevice/libimobiledevice-glue/pull/9
if we need to keep SHA-1 support in libimobiledevice for device support.

mexmer, 2022-09-07:
you cannot change signing for old devices, they will not understand and communicate with you. it's same issue, why we need to support SSL 3.0, when communicating with ios 1.0 devices. since we can't "patch" old system, to support new ciphers and algorithms, that were not available at time, they were manufactured.

either libimobile will support all apple devices (regardless age), and then you need to support all standards that were and are used, or it will support only "latest greatest", then it can adhere to modern "standards".

from security standpoint, there is no problem with supporting old ciphers and encryption algorithms, libimobile doesn't provide any service, it's just communication bridge, same for ubsmuxd, technically you can connect to it trough socket, but for that, usbmuxd can have communication socket use only latest, while for device communication, it can use all.

but there is issue, with what will be available in distribution, sooner or later you might find out, that libimobile compiled on latest version of your favorite distribution, cannot communicate with 6 years old apple devices, because libraries in your distribution do not have support for old encryption standards ...

hadess, 2022-09-07:
It's nice of you to answer, but I don't understand how some of it would answer the questions I asked.

> but there is issue, with what will be available in distribution, sooner or later you might find out, that libimobile compiled on latest version of your favorite distribution, cannot communicate with [...]

This is also what this issue is about, albeit 

SHA-2 family signatures already existed in 2010, so I'm not sure the comparison with SSL v3 getting deprecated is the right one. 

Re-pairing devices isn't a big deal, if the devices support other SHA-2 signatures.

> from security standpoint, there is no problem with supporting old ciphers and encryption algorithms, libimobile doesn't provide any service, it's just communication bridge, same for ubsmuxd, [...]

usbmuxd2 can talk over the network, so it's pretty important to know whether the use of SHA-1 can have an impact on the security/privacy of that traffic.

mexmer, 2022-09-07:
> It's nice of you to answer, but I don't understand how some of it would answer the questions I asked.
>
>
> This is also what this issue is about, albeit
>
> SHA-2 family signatures already existed in [...]
just because something existed, doesn't mean everyone implemeted it at time.
>
> Re-pairing devices isn't a big deal, if the devices support other SHA-2 signatures.
>
i can tell you, that older devices do not support it, i'm too lazy to check when apple started to support it, but definetly not before ios4
> usbmuxd2 can talk over the network, so it's pretty important to know whether the use of SHA-1 can have an impact on the security/privacy of that traffic.
well, i repeat then ... client can talk with usbmuxd2 with any moderm cipher and algorithm, so your network communication cab ne secure, but you still need to maintain compatibility (and therefore old standards supports) for communication with device. Wifi Sync is supported since ios 5, there you might be concerned with using old protocols on device, and for that will also make sense, if they support sha2 and tls 1.2
while SHA2 itself is more than 20 years old, and TLS 1.2 somewhere around 10 years, it was not before 2015 when SHA256 was addopted for TLS 1.2, and even then many devices didn't accept certificates with SHA256 signatures.

hadess, 2022-09-07:
> just because something existed, doesn't mean everyone implemeted it at time.

Knowing whether it was implemented was my question...

> > Re-pairing devices isn't a big deal, if the devices support other SHA-2 signatures.
> 
> i can tell you, that older devices do not support it, i'm too lazy to check when apple started to support it, but definetly not before ios4

That would mean implementing a fallback for:
- iPod touch 1st and 2nd gen ([source](https://en.wikipedia.org/wiki/IPod_Touch#Software))
- original iPhone ([source](https://en.wikipedia.org/wiki/IPhone#Software))
in terms of latest version of "iPhone OS" supported on those devices.

All the iPads support new enough versions ([source](https://en.wikipedia.org/wiki/IPad#Operating_system_support)).

> well, i repeat then ... client can talk with usbmuxd2 with any moderm cipher and algorithm, so your network communication cab ne secure, but you still need to maintain compatibility (and therefore [...]

SHA-256 was an example, there are other digests available in the SHA-2 suite which might have been implemented on the devices.

I'm not interested in knowing what usbmuxd2 "could" support, but rather how the code in libimobiledevice impacts usbmuxd2's ability to 1) use modern encryption and signatures when possible 2) still support older devices.

remyabel2, 2022-09-08:
FWIW, if I recall correctly, I remember a developer saying @mexmer is not a maintainer of this project. cc'ing @nikias.

mexmer, 2022-09-08:
well usbmuxd2 is not part of libimobile either. so discussing here, what can/need to be changed in usbmuxd2 here is somewhat pointless, but this information is still relevant for libimobile, because if distribution will indeed remove support for certain algorithms and protocol, it will loose compatibility with some devices.

remyabel2, 2022-09-08:
> well usbmuxd2 is not part of libimobile either. so discussing here, what can/need to be changed in usbmuxd2 here is somewhat pointless, 

They were providing an example.

> but this information is still relevant for libimobile, because if distribution will remove support for certain algorithms and protocols in their distributed crypto libraries, it will loose compatibility with some devices.

Hence the entire point of this discussion. I just think this conversation would be more productive with someone familiar with the codebase and can actually implement the challenges, rather than going around in circles.

mexmer, 2022-09-08:
> They were providing an example.
>
>
> Hence the entire point of this discussion. I just think this conversation would be more productive with someone familiar with the codebase and can actually [...]

i'm quite familiar with it because i'm working on commercial solution, that is partially based on libimobile, we have own implementation, we also need to ship own compiled openSSL library with SSL 3.0 enabled, to support old ios devices, when i will get to test bench with old devices (iphone 1st gen with ios2, ipad 1st gen with ios 4), i will  check, what they do accept, but those devices are not even able to connect to websites that require TLS 1.2

remyabel2, 2022-09-08:
> actually this discussion will be more relevant in libusbmuxd, because that's the client for communicating with usbmuxd service, and that's where you need to have actual communication protection, if it connects over network, and not over usb.

AFAIC the relevant crypto code is in libimobiledevice. The original issue is about distrusting SHA1 signatures.

> i'm quite familiar with it because i'm working on commercial solution, that is partially based on libimobile, we have own implementation, but i gathered lot of knowledge from libimobile code, we also ship own compiled openSSL library with SSL 3.0 enabled, to support old ios devices.

SSL 3.0 is insecure. If you have to patch OpenSSL to work with old devices for your proprietary solution, that's fine, not sure what that has to do with the issue at hand.

> when i will get access to test bench with old devices (iphone 1st gen with ios2, ipad 1st gen with ios 4), i will check, what they do accept, but those devices are not even able to connect to websites that require TLS 1.2

I don't know what this has to do with TLS either.

Again, I don't think this discussion is going anywhere. I would prefer to hear a response from a maintainer on the project. I'm going to be unsubscribing from this thread though as there isn't really anything else I can contribute at this point.

mexmer, 2022-09-09:
@remyabel2 you argumented with privacy, and why SHA2 is needed instead of SHA1, but you are blind to simple fact, that communication is not encrypted in first place most of the time. just because you throw SHA2 signature for limited ammount of payload, it will not make it more private and secure. yes, over the time, more and more stuff in device communication uses SSL connection, and there comes into play certificates, but that's just one piece of puzzle.

as for support of SHA2 signature in CA certificate on older devices, i will tell you next week, when i will have access to old devices. for now i can confirm it works with ios15 nad ios16 beta

remyabel2, 2022-09-09:
That's cool and I don't really care. Don't ping me, it resubscribes me to threads. Fedora is adopting a stronger crypto policy and libimobiledevice breaks because of the current crypto code. It's that simple. What *you* choose to do is not really relevant to me, a maintainer response would be more relevant as they would have a better response as to whether or not code changes could be implemented, and if not, what workarounds can be applied downstream. I don't understand why you show up in almost every thread arguing with users when you don't even work on the project.

nikias, 2022-09-10:
There is some code that uses SHA1 in `3rd_party/libsrp6a-sha512`, but I think it's only used when compiled against cryptolib or when no other supported crypto library is available. This code can technically be removed or replaced, since libimobiledevice only supports OpenSSL, mbedtls, or GnuTLS. I will see what I can do here.

Regarding SSL: If you need to support really old devices, I guess you just have to use an older OpenSSL version, since the newer versions will just not support SHA1. The code in libimobiledevice itself doesn't use SHA1 directly, it just tries to use an older TLS version. So there's nothing really that can be changed.

mexmer, 2022-09-12:
@nikias SHA1 is used directly in signature, when creating host and CA certificate, and upcoming policy for client will reject tls connection with sha1 signature, from my understanding, but when i thought about that from code perpective, while ssl connection uses certificates from pairing recods, maybe it's not libimobile concern anyways, because it registers callback to chain verification function.

still i tested what happens, if certificates are created with sha256 signature, and there was no problem on ios 15 and 16
when i did try on ios2, lockdown didn't connect in ssl mode ... so i assume it's missing sha256 signature support 
on ios4 it was working without issue ... don't have ios 3 device atm

if you look at it from this perpective, changing certificate generation function to use SHA256 or SHA1 depending on os version should do the trick. (but again, it might not be needed because of verification loopback)
https://github.com/libimobiledevice/libimobiledevice/blob/b5ce44449216195d02df386a1799efdaffe45a67/common/userpref.c#L422