libimobiledevice/libimobiledevice #1356: com.apple.os_trace_relay port for  libimobiledevice

Dantee296, 2022-09-09:
do we have **com.apple.os_trace_relay** service available for libimobiledevice
seems its already implemented in` [pymobiledevice3] (https://github.com/doronz88/pymobiledevice3)`

would be nice to have in libimobiledevice

https://github.com/doronz88/pymobiledevice3/blob/cec01585ab5fdbf204138ba0b7ea2f47e9262a54/pymobiledevice3/services/os_trace.py

Dantee296, 2022-09-12:
i tried implementing it but its not working ,

```
    plist_dict_set_item(dict, "Request", plist_new_string("PidList"));
      property_list_service_send_xml_plist(plistclient, dict);
```

able to sent request successfully
but `property_list_service_receive_plist` not working and return

`PROPERTY_LIST_SERVICE_E_SSL_ERROR`

@nikias please lmme know how to implement this service

mexmer, 2022-09-12:
@Dantee296 did you check source of src/syslog_relay.c and tools/idevicesyslog.c

syslog_relay is not property list service, and i suspect os_trace_relay is same, so you will need to construct connection with service factory, and use it directly.

check
https://github.com/libimobiledevice/libimobiledevice/blob/b5ce44449216195d02df386a1799efdaffe45a67/src/syslog_relay.c#L98

Dantee296, 2022-09-12:
@mexmer i did check that , seem syslog relay dsnt require any param/args after starting the service
where in os_relay we need to send plist ex . if we wanna get "running process list"

"Request" , "PidList"

i guess we need to implement it service layer bypassing plist send/receive 
probably by  ` "service_send" , "service_receive_with_timeout"`

mexmer, 2022-09-12:
actually i think it should be possible even now, i mean afc also communicates directly on socket you get from servicestart, i will look into it, when i have time, but atm. i don't have much free time to play with this stuf

mexmer, 2022-09-12:
i just did quick test in our application, setting up service link, after sending 
```
<?xml version="1.0" encoding="UTF-8"?>
<!DOCTYPE plist PUBLIC "-//Apple//DTD PLIST 1.0//EN"	"http://www.apple.com/DTDs/PropertyList-1.0.dtd">
<plist version="1.0">
<dict>
<key>Request</key>
<string>PidList</string>
</dict>
</plist>
```
it did receive back plist with process list, after skipping 1st byte (like that python library) i was able to decode it.

probly easiest will be to use regular send/receive, it accepts binary and xml form of plist, so you can easily push into socket PLIST in XML and await response.

Dantee296, 2022-09-12:
@mexmer can share sample code/snippet of how you were sending and receiving data

mexmer, 2022-09-12:
not really, we have own code, that is not using libimobile directly, it's abstraction, on top three different codes.
we are using libimobile only for communication with AMDS on windows, for direct communication we use our own communication layer, on top of apple windows ioctl driver.

but you should be able to use _sevice_send_ and _service_receive_ commands for what you need.
start service like syslog (using factory), then send data with _service_send_ a receive with _service_receice_

if you give me time, i might try to create gist for it ... no promise it will work tho

mexmer, 2022-09-13:
@nikias can you tell me, what's issue here?
i'm getting SERVICE_E_SSL_ERROR on service_receive call
https://gist.github.com/mexmer/6baa6be00bfb0d55f442e13704f27ecb

Dantee296, 2022-09-13:
@mexmer this is exact issue i was facing

Dantee296, 2022-09-13:
@mexmer i am receiving this from service response "Error_NoInitialMessage"

Dantee296, 2022-09-13:
i used your code but looking at pymobile code noticed he reading exact 4 bytes after redaing 1 byte before reading final msg. 

once i did that i recieved this message in 3rd rcv call

mexmer, 2022-09-13:
> i used your code but looking at pymobile code noticed he reading exact 4 bytes after redaing 1 byte before reading final msg.
> 
> once i did that i recieved this message in 3rd rcv call

ah this, recv_prefixed first reads 4 bytes, that are size of sent data, then reads amount, specified in those 4 bytes

mexmer, 2022-09-13:
of course i'm stupid ... too much abstractions leads to ... well people not seeing stuff.
service_send is direct send, so we need first send length of payload.

i fixed gist, so you can now get xml output of pidlist.
hope this helps.

Dantee296, 2022-09-14:
@mexmer  it working great now! 

can u also work on these two features ? and create pull for idevicetrace tool , it would be good addition in tools

// os trace logs

plist_dict_set_item(dict, "Request", plist_new_string("StartActivity"));
                    plist_dict_set_item(dict, "MessageFilter", plist_new_real(65535));
                    plist_dict_set_item(dict, "Pid", plist_new_real(-1));
                    plist_dict_set_item(dict, "StreamFlags", plist_new_real(60));
                    
         // create archive 

plist_dict_set_item(dict, "Request", plist_new_string("CreateArchive"));
                    plist_dict_set_item(dict, "SizeLimit", plist_new_real(size));
                    plist_dict_set_item(dict, "AgeLimit", plist_new_real(age));
                    plist_dict_set_item(dict, "StartTime", plist_new_real(start));

mexmer, 2022-09-15:
not now, but should not be hard to use gist as start, then do, what need to be done.
tbh. for proper serviceable tool, it would need to implemented same stuff, like is for syslog, including reader thread, and callbacks ... which i don't have time now to do.

closed 2022-09-15 by Dantee296