libimobiledevice/libimobiledevice #1357: [Question or suggestion] cmake or meson/ninja?

rubyFeedback, 2022-09-10:
I just ran into a problem where it was recommended to use "--disable-dependency-tracking"
by the configure script of libimobiledevice.

This led me indirectly to this question: do you guys consider using cmake and or meson/ninja
rather than configure?

It's not so important to change perhaps but I wanted to ask this anyway. I am slowly getting
tired of GNU configure scripts these days. It's also more annoying to deal with them on 
windows (I use Linux most of the time though).

xanadu-lee, 2022-12-20:
Hmm, it would be nice if there was a CMake configuration

mexmer, 2022-12-20:
i always disable cython bindings, i believe it was not updated for quite some time (and tbh. i never used it anyways), so i just avoid it, to not raise configuration and compilation errors.

xanadu-lee, 2022-12-21:
It may be that I am not good at learning, and I can't debug code well in MinGW and MSVC.
My main project is compiled with MSVC, and libimobiledevice is compiled with MinGW.
When something goes wrong, I currently can only try to block it, not solve it, because I can't track the problem very well.
It would be nice if there was a CMake configuration that would allow me to compile it with MSVC.

xanadu-lee, 2023-05-16:
When will we have a CMake configuration support?

yigithanyucedag, 2024-02-03:
> When will we have a CMake configuration support?

+1