libimobiledevice/libimobiledevice #1358: Heap use-after-free in restore.c

TrungNguyen1909, 2022-09-12:
libplist API states that `The caller should not free the returned node` for `plist_dict_get_item` see [here](https://github.com/libimobiledevice/libplist/blob/c3af449543795ad4d3ab178120ff69e90fdd2cc8/include/plist/plist.h#L385-L393).

However, `restored_get_value` incorrectly does so at [this line](https://github.com/libimobiledevice/libimobiledevice/blob/b5ce44449216195d02df386a1799efdaffe45a67/src/restore.c#L284-L291), which causes a use-after-free when the plist is freed.

Please see the ASAN report for more details

```
=================================================================
==93231==ERROR: AddressSanitizer: heap-use-after-free on address 0x000106d0cd70 at pc 0x000104a6b290 bp 0x00016b78a4b0 sp 0x00016b78a4a8
READ of size 8 at 0x000106d0cd70 thread T0
    #0 0x104a6b28c in plist_free_node plist.c:287
    #1 0x104a6b1b4 in plist_free_node plist.c:295
    #2 0x104a6b07c in plist_free plist.c:405
    #3 0x104838f48 in restored_client_free+0xb8 (libimobiledevice-1.0.6.dylib:arm64+0x1cf48)
    #4 0x10468b44c in restore_get_irecv_device+0xe0 (idevicerestore:arm64+0x10001744c)
    #5 0x104677bf0 in idevicerestore_start+0x164 (idevicerestore:arm64+0x100003bf0)
    #6 0x10467c378 in main+0x460 (idevicerestore:arm64+0x100008378)
    #7 0x1048dd088 in start+0x204 (dyld:arm64e+0x5088)

0x000106d0cd70 is located 32 bytes inside of 48-byte region [0x000106d0cd50,0x000106d0cd80)
freed by thread T0 here:
    #0 0x104e0fc94 in wrap_free+0x98 (libclang_rt.asan_osx_dynamic.dylib:arm64e+0x3fc94)
    #1 0x104839858 in restored_get_value+0xdc (libimobiledevice-1.0.6.dylib:arm64+0x1d858)
    #2 0x10468b440 in restore_get_irecv_device+0xd4 (idevicerestore:arm64+0x100017440)
    #3 0x104677bf0 in idevicerestore_start+0x164 (idevicerestore:arm64+0x100003bf0)
    #4 0x10467c378 in main+0x460 (idevicerestore:arm64+0x100008378)
    #5 0x1048dd088 in start+0x204 (dyld:arm64e+0x5088)

previously allocated by thread T0 here:
    #0 0x104e0ff24 in wrap_calloc+0x9c (libclang_rt.asan_osx_dynamic.dylib:arm64e+0x3ff24)
    #1 0x104a75e14 in node_create+0x20 (libplist-2.0.3.dylib:arm64+0x51e14)
    #2 0x104a69700 in plist_new_node plist.c:217
    #3 0x104a357f0 in node_from_xml xplist.c:1140
    #4 0x104a33c14 in plist_from_xml xplist.c:1480
    #5 0x104824330 in internal_plist_receive_timeout+0x458 (libimobiledevice-1.0.6.dylib:arm64+0x8330)
    #6 0x1048243f8 in property_list_service_receive_plist+0x20 (libimobiledevice-1.0.6.dylib:arm64+0x83f8)
    #7 0x104839258 in restored_receive+0x60 (libimobiledevice-1.0.6.dylib:arm64+0x1d258)
    #8 0x1048393a0 in restored_query_type+0xc4 (libimobiledevice-1.0.6.dylib:arm64+0x1d3a0)
    #9 0x10468b3c0 in restore_get_irecv_device+0x54 (idevicerestore:arm64+0x1000173c0)
    #10 0x104677bf0 in idevicerestore_start+0x164 (idevicerestore:arm64+0x100003bf0)
    #11 0x10467c378 in main+0x460 (idevicerestore:arm64+0x100008378)
    #12 0x1048dd088 in start+0x204 (dyld:arm64e+0x5088)

SUMMARY: AddressSanitizer: heap-use-after-free plist.c:287 in plist_free_node
Shadow bytes around the buggy address:
  0x007020dc1950: fa fa fd fd fd fd fd fd fa fa 00 00 00 00 00 00
  0x007020dc1960: fa fa fd fd fd fd fd fd fa fa fd fd fd fd fd fd
  0x007020dc1970: fa fa 00 00 00 00 00 00 fa fa 00 00 00 00 00 00
  0x007020dc1980: fa fa fd fd fd fd fd fd fa fa 00 00 00 00 00 00
  0x007020dc1990: fa fa 00 00 00 00 00 00 fa fa fd fd fd fd fd fd
=>0x007020dc19a0: fa fa fd fd fd fd fd fd fa fa fd fd fd fd[fd]fd
  0x007020dc19b0: fa fa fd fd fd fd fd fd fa fa 00 00 00 00 00 00
  0x007020dc19c0: fa fa fd fd fd fd fd fd fa fa fd fd fd fd fd fd
  0x007020dc19d0: fa fa fd fd fd fd fd fd fa fa fd fd fd fd fd fd
  0x007020dc19e0: fa fa fd fd fd fd fd fd fa fa fd fd fd fd fd fa
  0x007020dc19f0: fa fa fd fd fd fd fd fd fa fa fd fd fd fd fd fd
Shadow byte legend (one shadow byte represents 8 application bytes):
  Addressable:           00
  Partially addressable: 01 02 03 04 05 06 07
  Heap left redzone:       fa
  Freed heap region:       fd
  Stack left redzone:      f1
  Stack mid redzone:       f2
  Stack right redzone:     f3
  Stack after return:      f5
  Stack use after scope:   f8
  Global redzone:          f9
  Global init order:       f6
  Poisoned by user:        f7
  Container overflow:      fc
  Array cookie:            ac
  Intra object redzone:    bb
  ASan internal:           fe
  Left alloca redzone:     ca
  Right alloca redzone:    cb
```

pekn, 2022-09-14:
How did you enable address sanitizer to libimobiledevice libraries?
Just wondering, would be useful to know..

nikias, 2022-09-14:
@TrungNguyen1909 thanks for catching this, fixed with ef7cf8eb545d9ed3fb1f351376dec71608d0127b.

@pekn basically you can edit `configure.ac` and e.g. add `-fsanitize=address` to this line (within the quotes):
```
AS_COMPILER_FLAGS(GLOBAL_CFLAGS, "-Wall -Wextra -Wmissing-declarations -Wredundant-decls -Wshadow -Wpointer-arith -Wwrite-strings -Wswitch-default -Wno-unused-parameter -fsigned-char -fvisibility=hidden")
```

closed 2022-09-14 by TrungNguyen1909