libimobiledevice/libimobiledevice #1361: ErrorCode 104: Multiple errors uploading files (MBErrorDomain/104)

kevinant2, 2022-09-18:
Hi All,

Just would like to know if this is solvable or if there are any ways around this. I've encountered backup issues as the following:
- The backup finished uploading all the files
- Just before completion of the backup process, the error "ErrorCode 104: Multiple errors uploading files (MBErrorDomain/104)" appears. 
- Status.plist, info.plist generated,
- However, Manifest.plist and Manifest.db are not generated.

If there are no solution, are there any way to get Manifest.plist and Manifest.db generated? 

Thank you very much for your help, the following is the output from the libimobiledevice

Kevin

Receiving files
[==================================================] 100% (803.2 KB/803.2 KB)
[==================================================]  99% Finished
Receiving files
[==================================================]  99% Finished/41.0 MB)
Receiving files
[==================================================] 100% (376.3 KB/376.3 KB)
[==================================================]  99% Finished
ErrorCode 104: Multiple errors uploading files (MBErrorDomain/104)
Received 193946 files from device.
Backup Failed (Error Code 104).

mexmer, 2022-09-19:
if you see this error, there must have been some errors during backup already, either phone was not able to read file from device, or idevicebackup didn't save the file, but responded with error back to device (might be permission issue, or not enough free space ... but you should see that error).

manifest is generated by device, and it will not be generated in case of incomplete backup.

airwolf2026, 2022-09-19:
May be you could delete some app or data on device, then try again.

This is the device backup service problem.

mexmer, 2022-09-19:
yes that is also possibly, backup process (inside device) makes copy of files (at least opened ones) before sending them to computer, if there is not enough room in device, it will internally error, and not send file, but sometimes continue until end.

kevinant2, 2022-09-20:
Hi all,

thanks for your responses. 

This happens originally with the computer sleeping in the middle of the backup process sometimes ago. Then "Null" photo album related files were generated. Those "Null" files can be moved when filesystem is mounted but they can't be deleted and it appears to be quite weird. 

Initially the phone would not able to upload any files (it was on iOS 14.6) and showing error that Null bytes were found in some files. After upgrading it to iOS15.3 the backup progress proceeds and it appears that almost all of files are uploaded before the error occurs. The Manifest.db and manifest.plist were not generated in the backup.

So I just wondering if the Manifest can be generated in some way or if those "Null" files can be removed. 

Thanks again.

Kevin

mexmer, 2022-09-20:
unfortunately, unless you fix data inside phone, nothing can be done.

manifest.plist and manifest.db are generated by phone (backupservice in phone), and it will not create them, until it's finished.
manifest database contains files metadata as well as their encryption keys, and those are stored inside phone, and i don't know about any way how to get them (maybe from jailbroken phone, but i don't know about working jailbreak for ios 15.3 either).

kevinant2, 2022-09-22:
Thanks mexmer, 

Do you know if there are any way to effectively delete those null byte files on the iphone? They are undeletable with current tools.

Thanks,

Kevin

mexmer, 2022-09-22:
do you have in target folder also previous backup of phone? if so, then do backup to different folder, because normaly phone cheks for existing backup, and then only upload changed files. if there is no previous backup, it does full backup.

kevinant2, 2022-09-25:
Thanks mexmer, I don't have a previous backup of this phone so it's doing full backup, therefore I am guessing it's touching those generated NULL files.

kevinant2, 2022-09-25:
Thanks mexmer, Nope I don't have a target folder of previous backup of the phone, so it does full back up. It would be great if anyone know a way to get the phone perform the backup bypassing those null files. 

I've recorded iphone path and the SHA1 hash of the file but there are still gaps between that and the manifest file.

mexmer, 2022-09-25:
can you post syslog?
start on one console idevicesyslog, and redirect it's output to file, then on other console start idevicebackup.
stop syslog, after it fails.

if you are concerned about privacy, just send me download link for log trough private message on github.

kevinant2, 2022-10-01:
Thanks Mexmer. 

The following is the syslog around the time when error occured. It appears that there's a "Backup Completed" message sent before the error is thrown out . 

Hope this will gives some clue. 

Oct  1 17:58:33 kernel[0] <Notice>: tcp_timers: tcp_output() returned 0 with retransmission timer disabled for 51179 > 51180 in state 4, reset timer to 28850
Oct  1 17:58:33 BackupAgent2[46806] <Notice>: Notified MBiTunesStorePlugin of backingUpSQLiteFileCopyAtPath:temporaryPath: in 0.000s
Oct  1 17:58:33 BackupAgent2[46806] <Notice>: Notified MBiTunesStorePlugin of backingUpSQLiteFileCopyAtPath:temporaryPath: in 0.000s
Oct  1 17:58:33 BackupAgent2[46806] <Notice>: Notified MBiTunesStorePlugin of backingUpSQLiteFileCopyAtPath:temporaryPath: in 0.000s
Oct  1 17:58:33 BackupAgent2[46806] <Notice>: Notified MBiTunesStorePlugin of backingUpSQLiteFileCopyAtPath:temporaryPath: in 0.000s
Oct  1 17:58:33 BackupAgent2[46806] <Notice>: Notified MBiTunesStorePlugin of backingUpSQLiteFileCopyAtPath:temporaryPath: in 0.000s
Oct  1 17:58:33 BackupAgent2[46806] <Notice>: Notified MBiTunesStorePlugin of backingUpSQLiteFileCopyAtPath:temporaryPath: in 0.000s
Oct  1 17:58:33 BackupAgent2[46806] <Notice>: Notified MBiTunesStorePlugin of backingUpSQLiteFileCopyAtPath:temporaryPath: in 0.000s
Oct  1 17:58:34 mediaserverd(CMCapture)[8126] <Notice>: m11:31232, m12:0, m21:413968, m22:55008, m23:410128, m24:386208
Oct  1 17:58:34 kernel(AppleBCMWLANCore)[0] <Notice>: LQM-WiFi:BT Coex Stats3:btDur=19ms(0%) status=0x23 reqTypeMap=0x40 req=2 gnt=1 abort=10642 rx1ovfl=3355 latency=848 sucPM=1 sucCTS2S=0 wlanTXPreempt=0 wlanRXPreempt=43 apTXafterPM=0 gntAudio=0 denyAudio=0 gntA2DP=0 denyA2DP=0 gntSniff=0 denySniff=0 [5057ms]
Oct  1 17:58:34 kernel(AppleBCMWLANCore)[0] <Notice>: LQM-WiFi: ccmpreplay=12 ccmpreplay_qosdata_nobapol_rxretry=0
Oct  1 17:58:34 kernel(IO80211Family)[0] <Notice>: postMessageInternal:isPipeOpened:1, msg 175, dataLen 140
Oct  1 17:58:34 kernel(IO80211Family)[0] <Notice>: postMessageInternal:isPipeOpened:1, msg 155, dataLen 12
Oct  1 17:58:34 kernel[0] <Notice>: tcp_timers: tcp_output() returned 0 with retransmission timer disabled for 51179 > 51180 in state 4, reset timer to 1763
Oct  1 17:58:34 kernel[0] <Notice>: tcp_timers: tcp_output() returned 0 with retransmission timer disabled for 51179 > 51180 in state 4, reset timer to 304
Oct  1 17:58:35 kernel[0] <Notice>: tcp_timers: tcp_output() returned 0 with retransmission timer disabled for 51179 > 51180 in state 4, reset timer to 540
Oct  1 17:58:35 kernel[0] <Notice>: tcp_timers: tcp_output() returned 0 with retransmission timer disabled for 51179 > 51180 in state 4, reset timer to 373
Oct  1 17:58:35 BackupAgent2[46806] <Notice>: Notified MBiTunesStorePlugin of backingUpSQLiteFileCopyAtPath:temporaryPath: in 0.000s
Oct  1 17:58:35 kernel[0] <Notice>: tcp_timers: tcp_output() returned 0 with retransmission timer disabled for 51179 > 51180 in state 4, reset timer to 307
Oct  1 17:58:35 mediaanalysisd-service[46954] <Notice>: Media Analysis task progress: 31.62%
Oct  1 17:58:35 mediaserverd(CMCapture)[8126] <Notice>: m11:31248, m12:0, m21:414032, m22:53216, m23:409712, m24:385952
Oct  1 17:58:35 kernel[0] <Notice>: tcp_timers: tcp_output() returned 0 with retransmission timer disabled for 51179 > 51180 in state 4, reset timer to 428
Oct  1 17:58:35 kernel[0] <Notice>: tcp_timers: tcp_output() returned 0 with retransmission timer disabled for 51179 > 51180 in state 4, reset timer to 365
Oct  1 17:58:35 kernel[0] <Notice>: tcp_timers: tcp_output() returned 0 with retransmission timer disabled for 51179 > 51180 in state 4, reset timer to 457
Oct  1 17:58:35 BackupAgent2[46806] <Error>: Failed to upload after 1518 batches, s:{0,0,0,0}, t:20381.676, r:0.000 - Error Domain=MBErrorDomain Code=104 UserInfo={NSLocalizedDescription=<private>}
Oct  1 17:58:35 BackupAgent2[46806] <Notice>: Finished backup in 20622.153s
Oct  1 17:58:35 BackupAgent2[46806] <Error>: Failed to backup: Error Domain=MBErrorDomain Code=104 UserInfo={NSLocalizedDescription=<private>}
Oct  1 17:58:36 BackupAgent2[46806] <Notice>: ManifestDB: closing <private>
Oct  1 17:58:36 kernel(IO80211Family)[0] <Notice>: LQM-WIFI-CT: ofdmDesense=0 bphyDesense=0 reason=0x0
Oct  1 17:58:36 BackupAgent2[46806] <Notice>: Notified MBRestoreAppsPlugin of endedBackupWithEngine:error: in 0.002s
Oct  1 17:58:36 BackupAgent2[46806] <Notice>: Notified MBLockdownPlugin of endedBackupWithEngine:error: in 0.001s
Oct  1 17:58:36 lockdownd[75] <Notice>: _disable_sigpipe: <private>
Oct  1 17:58:36 kernel(AppleBCMWLANCore)[0] <Notice>: LQM-WiFi: ccmpreplay=12 ccmpreplay_qosdata_nobapol_rxretry=0
Oct  1 17:58:36 kernel(IO80211Family)[0] <Notice>: postMessageInternal:isPipeOpened:1, msg 155, dataLen 12
Oct  1 17:58:36 BackupAgent2[46806] <Notice>: Notified MBAppleCarePlugin of endedBackupWithEngine:error: in 0.034s
Oct  1 17:58:36 BackupAgent2[46806] <Notice>: Notified MBAnalyticsEventPlugin of endedBackupWithEngine:error: in 0.005s
Oct  1 17:58:36 BackupAgent2[46806] <Notice>: Unmounting APFS snapshot <private> at /.ba and removing it
Oct  1 17:58:36 mediaserverd(CMCapture)[8126] <Notice>: m11:31248, m12:0, m21:414128, m22:60512, m23:409440, m24:380672
Oct  1 17:58:36 kernel(apfs)[0] <Notice>: apfs_log_mount_unmount:1828: disk0s1s2 unmounting volume com.apple.mobilebackup-iTunesBackup, requested by: BackupAgent2 (pid 46806); parent: launchd (pid 1)
[... cut ...]
Oct  1 17:58:36 kernel(apfs)[0] <Notice>: cleanup_snapshot_purgatory:1619: disk0s1s2 processing snap xid 12278237...
Oct  1 17:58:36 BackupAgent2(MobileBackup)[46806] <Notice>: Deleted snapshot com.apple.mobilebackup-iTunesBackup at /private/var
Oct  1 17:58:36 BackupAgent2(MobileBackup)[46806] <Notice>: Deleted snapshot com.apple.mobilebackup-iTunesBackup
Oct  1 17:58:36 kernel(apfs)[0] <Notice>: delete_clone_fs:2627: disk0s1s2 forward merge, src_snap_xid 12278237, dest_snap_xid 0, cur_tree_count 810634, next_tree_count 13312

mexmer, 2022-10-01:
i did really hope, it will show some file name, but doesn't seems so, also log entry looks it has problem backing up itunes media database, it did send notification to itunes that it should copy media database to temporary folder, and after few tries it failed. 

maybe itunes in phone is downloading something, no idea, it's blind shot, but put phone into airplane mode, then do backup.

kevinant2, 2022-10-02:
Thanks mexmer, 
I have more complete log but it's more than half a million of lines. 
I've just tried to backup with the phone in the airplane mode but still with same MBErrorDomain/104. 
The following is the log with the new backup. The log is as follows:

Oct  3 06:47:18 mediaanalysisd-service(PhotoLibraryServices)[47898] <Notice>: resumeChangeDistribution: _observerCount became 1 (resuming), _lastToken = <NSPersistentHistoryToken - {
    "AC005BFE-9A5F-49F5-852D-531E057F3AA8" = 118028;
}>
Oct  3 06:47:18 mediaserverd(CMCapture)[8126] <Notice>: m11:31264, m12:0, m21:375520, m22:76032, m23:346016, m24:599840
Oct  3 06:47:18 kernel(IO80211Family)[0] <Notice>: LQM-WIFI-CT: ofdmDesense=0 bphyDesense=0 reason=0x0
Oct  3 06:47:19 mediaserverd(CMCapture)[8126] <Notice>: m11:31264, m12:0, m21:375520, m22:72608, m23:346000, m24:601616
Oct  3 06:47:20 mediaserverd(CMCapture)[8126] <Notice>: m11:31232, m12:0, m21:375232, m22:73360, m23:331488, m24:605344
Oct  3 06:47:20 BackupAgent2[47278] <Notice>: Notified MBiTunesStorePlugin of backingUpSQLiteFileCopyAtPath:temporaryPath: in 0.000s
Oct  3 06:47:20 mediaanalysisd-service(PhotoLibraryServices)[47898] <Notice>: pauseChangeDistribution: _observerCount became 0 (pausing), _lastToken = <NSPersistentHistoryToken - {
    "AC005BFE-9A5F-49F5-852D-531E057F3AA8" = 118028;
}>
Oct  3 06:47:20 mediaanalysisd-service(PhotoLibraryServices)[47898] <Notice>: resumeChangeDistribution: _observerCount became 1 (resuming), _lastToken = <NSPersistentHistoryToken - {
    "AC005BFE-9A5F-49F5-852D-531E057F3AA8" = 118028;
}>
Oct  3 06:47:21 BackupAgent2[47278] <Notice>: Notified MBiTunesStorePlugin of backingUpSQLiteFileCopyAtPath:temporaryPath: in 0.000s
Oct  3 06:47:21 BackupAgent2[47278] <Notice>: Notified MBiTunesStorePlugin of backingUpSQLiteFileCopyAtPath:temporaryPath: in 0.000s
Oct  3 06:47:21 kernel[0] <Notice>: tcp_timers: tcp_output() returned 0 with retransmission timer disabled for 51255 > 51256 in state 4, reset timer to 25184
Oct  3 06:47:21 BackupAgent2[47278] <Notice>: Notified MBiTunesStorePlugin of backingUpSQLiteFileCopyAtPath:temporaryPath: in 0.000s
Oct  3 06:47:21 BackupAgent2[47278] <Notice>: Notified MBiTunesStorePlugin of backingUpSQLiteFileCopyAtPath:temporaryPath: in 0.000s
Oct  3 06:47:21 BackupAgent2[47278] <Notice>: Notified MBiTunesStorePlugin of backingUpSQLiteFileCopyAtPath:temporaryPath: in 0.000s
Oct  3 06:47:21 BackupAgent2[47278] <Notice>: Notified MBiTunesStorePlugin of backingUpSQLiteFileCopyAtPath:temporaryPath: in 0.000s
Oct  3 06:47:21 mediaserverd(CMCapture)[8126] <Notice>: m11:31232, m12:0, m21:375264, m22:83520, m23:323280, m24:605296
Oct  3 06:47:21 BackupAgent2[47278] <Notice>: Notified MBiTunesStorePlugin of backingUpSQLiteFileCopyAtPath:temporaryPath: in 0.000s
Oct  3 06:47:21 kernel(AppleBCMWLANCore)[0] <Notice>: LQM-WiFi: ccmpreplay=12 ccmpreplay_qosdata_nobapol_rxretry=0
Oct  3 06:47:21 kernel(IO80211Family)[0] <Notice>: postMessageInternal:isPipeOpened:1, msg 175, dataLen 140
Oct  3 06:47:22 mediaserverd(CMCapture)[8126] <Notice>: m11:31232, m12:0, m21:375232, m22:82368, m23:324608, m24:605360
Oct  3 06:47:22 kernel[0] <Notice>: tcp_timers: tcp_output() returned 0 with retransmission timer disabled for 51255 > 51256 in state 4, reset timer to 1713
Oct  3 06:47:22 kernel[0] <Notice>: tcp_timers: tcp_output() returned 0 with retransmission timer disabled for 51255 > 51256 in state 4, reset timer to 406
Oct  3 06:47:22 kernel[0] <Notice>: tcp_timers: tcp_output() returned 0 with retransmission timer disabled for 51255 > 51256 in state 4, reset timer to 381
Oct  3 06:47:22 kernel[0] <Notice>: tcp_timers: tcp_output() returned 0 with retransmission timer disabled for 51255 > 51256 in state 4, reset timer to 473
Oct  3 06:47:22 kernel[0] <Notice>: tcp_timers: tcp_output() returned 0 with retransmission timer disabled for 51255 > 51256 in state 4, reset timer to 410
Oct  3 06:47:23 BackupAgent2[47278] <Notice>: Notified MBiTunesStorePlugin of backingUpSQLiteFileCopyAtPath:temporaryPath: in 0.000s
Oct  3 06:47:23 kernel[0] <Notice>: tcp_timers: tcp_output() returned 0 with retransmission timer disabled for 51255 > 51256 in state 4, reset timer to 333
Oct  3 06:47:23 kernel[0] <Notice>: tcp_timers: tcp_output() returned 0 with retransmission timer disabled for 51255 > 51256 in state 4, reset timer to 398
Oct  3 06:47:23 mediaserverd(CMCapture)[8126] <Notice>: m11:31232, m12:0, m21:375248, m22:89840, m23:324080, m24:596160
Oct  3 06:47:23 kernel(IO80211Family)[0] <Notice>: LQM-WIFI-CT: ofdmDesense=0 bphyDesense=0 reason=0x0
Oct  3 06:47:24 mediaserverd(CMCapture)[8126] <Notice>: m11:31232, m12:0, m21:375264, m22:120304, m23:324064, m24:566144
Oct  3 06:47:24 kernel[0] <Notice>: tcp_timers: tcp_output() returned 0 with retransmission timer disabled for 51255 > 51256 in state 4, reset timer to 1791
Oct  3 06:47:24 kernel[0] <Notice>: tcp_timers: tcp_output() returned 0 with retransmission timer disabled for 51255 > 51256 in state 4, reset timer to 473
Oct  3 06:47:25 mediaserverd(CMCapture)[8126] <Notice>: m11:31280, m12:0, m21:375232, m22:120544, m23:326320, m24:566240
Oct  3 06:47:25 BackupAgent2[47278] <Error>: Failed to upload after 1518 batches, s:{0,0,0,0}, t:50355.666, r:0.000 - Error Domain=MBErrorDomain Code=104 UserInfo={NSLocalizedDescription=<private>}
Oct  3 06:47:25 BackupAgent2[47278] <Notice>: Finished backup in 50597.004s
[... cut ...]
Oct  3 06:47:26 kernel(apfs)[0] <Notice>: fs_insert_snapshot_metadata:265: disk0s1s2 snap_name='com.apple.apfs.purgatory.bbcfed' snap_xid 12308461 extentref oid 61865521 sblock oid 61868754
Oct  3 06:47:26 kernel(apfs)[0] <Notice>: nx_global_extent_manipulation_unlock:5465: disk0s1 nx_global_extent_manipulation_unlock(0, 1), (0, 0, 1, 1)
Oct  3 06:47:26 kernel(apfs)[0] <Notice>: cleanup_snapshot_purgatory:1619: disk0s1s2 processing snap xid 12308461...
Oct  3 06:47:26 kernel(apfs)[0] <Notice>: delete_clone_fs:2627: disk0s1s2 forward merge, src_snap_xid 12308461, dest_snap_xid 0, cur_tree_count 816349, next_tree_count 16534

mexmer, 2022-10-03:
do you have any mac or windows pc with itunes, so you can try to do local backup on them?

kevinant2, 2022-11-06:
Hi mexmer,  I've tried to perform a backup with itunes / airplane mode. The backup files are there but the same issue still occurs and there are no manifest.db.

mexmer, 2022-11-07:
if even desktop itunes fails, then definetly something is wrong either with database, or filesystem itself.
does syslog show same multiple _MBiTunesStorePlugin backingUpSQLiteFileCopyAtPath:temporaryPath_ message?

if so, then probly best would be to do icloud backup, prior doing any experiments on filesystem, but then i would mount device filesystem and delete photos database, and do idevicebackup, if this doesn't work, second step would be deleting itunes media database. but tbh. deleting anything from phone filesystem direcly can also cause another bunch of issues, and also data loss (by deleting photos database, you will loose albums, by deleting itunes media database, you will loose list of synced media, and you might have then orphaned media in itunes folder, that will not show up, but will take space). it's not a thing i would recommend doing at all.

samuelsadok, 2024-03-21:
As others have pointed out, this is not a libimobiledevice issue. I ran into the same issue with a normal backup in macOS Finder. But since I came across this thread while investigating, I figured I'd add my findings here in case it helps anyone else.

**Your device needs to be jailbroken for this.**

In my case, on a iOS 15.8.2 device, I could see a handful of messages like these in the syslog:
```
BackupAgent2	Failed to upload <private> to <private> in batch 679: <NSError:0x281101f50(MBErrorDomain:100) - {NSLocalizedDescription = <private>;}>
BackupAgent2	Failed to upload <private> to <private> in batch 707: <NSError:0x281120cc0(MBErrorDomain:100) - {NSLocalizedDescription = <private>;}>
...
BackupAgent2	Failed to upload after X batches [...]
```
The file names are redacted behind `<private>` but could be revealed (in a second run) thanks to @EthanArbuckle's method described here: https://github.com/EthanArbuckle/unredact-private-os_logs

```
378	error	16:44:14.650227+0100	BackupAgent2	Failed to upload /.ba/mobile/Media/DCIM/108APPLE/IMG_8720.HEIC to ea75b4bce2c5c15387720dc3202373ac2dae53c5/Snapshot/14/1477d34c9071bb8760c8d25d182563029539a424 in batch 679: <NSError:0x281101f50(MBErrorDomain:100) - {NSLocalizedDescription = "Input/output error (5)";}>
378	error	16:47:36.416450+0100	BackupAgent2	Failed to upload /.ba/mobile/Media/DCIM/119APPLE/IMG_9294.JPG to ea75b4bce2c5c15387720dc3202373ac2dae53c5/Snapshot/74/74bf5698e8ef46ea869a31a9579f65d241f5acd1 in batch 707: <NSError:0x281120cc0(MBErrorDomain:100) - {NSLocalizedDescription = "Input/output error (5)";}>
...
```
This is just an exert, the bad files also included application data, not only media files.

So I ran a backup while recording the syslog to get a list of all bad files. (I don't know why they were bad, but this device had update problems before so maybe that was related.) Then I moved all bad files to a stash location. (I prefer moving over deleting, in case I need to revert it.) Note that `/.ba/` is a snapshot mount (probably) and the original files are in `/private/var/`.

```sh
mkdir /private/var/backup_quarantine
mv /private/var/mobile/Media/DCIM/108APPLE/IMG_8720.HEIC /private/var/backup_quarantine/
mv /private/var/mobile/Media/DCIM/119APPLE/IMG_9294.JPG /private/var/backup_quarantine/
...
```

And that's it, after that the backup completed successfully!

### Good backup log

Just for reference, here's the syslog of a **good** backup (filter: Errors&Warnings, "BackupAgent2"). This was recorded on a different, normally functioning iOS 15.8 device. This helped me distinguish true errors from red herrings:

```
19819	error	16:04:10.832951+0100	BackupAgent2	[ICUserIdentityStore] - initializing account histories with <private>
19819	error	16:04:10.878356+0100	BackupAgent2	Missing books entry.
19819	error	16:04:55.213037+0100	BackupAgent2	unmount failed at /.ba: [22: Invalid argument]
19819	error	16:04:56.412266+0100	BackupAgent2	API call with NULL database connection pointer
19819	error	16:04:56.412297+0100	BackupAgent2	misuse at line 176255 of [9ff244ce07]
19819	error	16:04:56.412322+0100	BackupAgent2	can't clear busy handler
19819	error	16:04:57.417016+0100	BackupAgent2	open_dprotected_np failed at <private>: [2: No such file or directory]
19819	error	16:04:57.417727+0100	BackupAgent2	open_dprotected_np failed at <private>: [2: No such file or directory]
[repeats several times...]
19819	error	16:06:46.756053+0100	BackupAgent2	cannot open file at line 37488 of [9ff244ce07]
19819	error	16:06:52.266576+0100	BackupAgent2	cannot open file at line 37488 of [9ff244ce07]

19819	error	16:07:03.331610+0100	BackupAgent2	no such collation sequence: IDBKEY in "CREATE TABLE Records (objectStoreID INTEGER NOT NULL ON CONFLICT FAIL, key TEXT COLLATE IDBKEY NOT NULL ON CONFLICT FAIL, value NOT NULL ON CONFLICT FAIL, recordID INTEGER PRIMARY KEY)"
19819	error	16:07:03.332194+0100	BackupAgent2	_sqlite3_db_copy_compact failed at <private>: 1/0x1
19819	error	16:07:03.332593+0100	BackupAgent2	Failed to compact SQLite database at <private>: Error Domain=MBErrorDomain Code=16 UserInfo={NSFilePath=<private>, NSLocalizedDescription=<private>}
[group of 3 warnings occurs several times]

19819	error	16:08:40.692804+0100	BackupAgent2	unmount failed at /.ba: [22: Invalid argument]
19819	error	16:08:40.716527+0100	BackupAgent2	unmount failed at /.ba: [22: Invalid argument]
```

### Manifest.plist, Manifest.db

Since there were some questions about these files:
- Manifest.plist is generated on the iDevice during the initial phase of a backup.
- Manifest.db is created on the iDevice early in the backup and grows in size as the backup proceeds.

On a jailbroken device you can see both files while the backup is running (UUID in the path may vary):

```sh
> ls -lah /private/var/root/Library/Caches/Backup/BC2368F2-E882-480B-A338-FD1167735347/
total 396384
drwxr-xr-x   6 root  wheel   192B Mar 21 16:24 .
drwxr-xr-x  13 root  wheel   416B Mar 21 16:21 ..
-rw-r--r--   1 root  wheel   187M Mar 21 16:51 Manifest.db
-rw-r--r--   1 root  wheel    64K Mar 21 16:22 Manifest.db-shm
-rw-r--r--   1 root  wheel   1.2M Mar 21 16:52 Manifest.db-wal
-rw-r--r--   1 root  wheel   193K Mar 21 16:24 Manifest.plist
```

This folder is deleted at the end of the backup. So an alternative approach to what I described above could be to patch `/usr/libexec/BackupAgent2` or use some filesystem trickery to prevent deletion at the end of a backup and then transfer them manually (e.g. via SSH). You'd probably need to use additional tools to repair the resulting backup.

j2048b, 2026-10-20:
Im not sure how to do any of this but im on ios 27.0.1 iphone 13 pro max and having a hard time myself… this thing wont back up to icloud or a new mac laptop either… i had this same error and need help fixing this… do i need to be jailbroken? If so i need to find out how to do that then how to fix these log files… ive tried 3rd party apps to do a back up and each one fails as well…  error code 104 every time different apps… i dont want to buy a new phone i just want to back everything up so i can reset this dang phone…

kevinant2, 2026-10-31:
> requests on the go with GitHub Mobile for iOS
> and Android
> Download it today!
> <libimobiledevice/libimobiledevice/issues/1361/6283245509

Hi J2048b,

It has been a quite long time.

I have resolved this by the following steps

a. record all output from MGMINW32.in a text file
b. output will show file path of the iphone to be backed up. I've wrote
python script to collect all of them
c. wrote a python script to create Manifest.db and Manifest.plist manually.
The backup filename is SHA-256 of the file path, so you just match the
filename generated with the SHA-256 of filepath.
d. bought https://deciphertools.com/ (Decipher backup) and had their
engineer to fix the Manifest.db and Manifest.plist

Cheers,

Kevin

j2048b, 2026-10-31:
> 18\. Hi J2048b,
>      
>      It has been a quite long time.
>      
>      I have resolved this by the following steps
>      
>      a. record all output from MGMINW32.in a text file
>      b. output will show file path of the iphone to be backed up. I've wrote
>      python script to collect all of them
>      c. wrote a python script to create Manifest.db and Manifest.plist manually.
>      The backup filename is SHA-256 of the file path, so you just match the
>      filename generated with the SHA-256 of filepath.
>      d. bought https://deciphertools.com/ (Decipher backup) and had their
>      engineer to fix the Manifest.db and Manifest.plist
>      

Thats awesome thanks! I cant code or write python scripts so i wasnt sure if there was a piece of software that would assist me or help me in grabbing sll this stuff and then i could get the software u showed to help fix the remaining issues…

I was told by apple to erase my phone and start over but im not doing that, thats their way of not fixing the issue… their engineers told me it was a nand error which is weird but possible i suppose?

kevinant2, 2026-11-01:
Hi J2048b,

It has been a while ago and I don't think it would be too hard to reverse
engineer Manifest.db/ Manifest.Plist and create one using python.

As long as you have all output (could be millions of rows) from Mgminw32
when you use libimobiledevice for backup, you would have files and the
outputs which you can recreate Manifest.db .

Even so, your backup file would contain data etc. so I don't think you need
to reset this. This usually happens when NAND is over capacity.

Cheers,

Kevin

j2048b, 2026-11-01:
> 20\. Hi J2048b,
>      
>      It has been a while ago and I don't think it would be too hard to reverse
>      engineer Manifest.db/ Manifest.Plist and create one using python.
>      
>      As long as you have all output (could be millions of rows) from Mgminw32
>      when you use libimobiledevice for backup, you would have files and the
>      outputs which you can recreate Manifest.db .
>      
>      Even so, your backup file would contain data etc. so I don't think you need
>      to reset this. This usually happens when NAND is over capacity.

Thank you! Ill give it a go!