libimobiledevice/libimobiledevice #1364: iOS16.0 cannot install developer plugin

xanadu-lee, 2022-09-19:
I upgraded a device from 15.6 to 16.0 and then the program that installed the developer plugin did not work successfully.

~~~
D:\libimobiledevice>ideviceimagemounter.exe -u 00008110-001628DA1EDB801E -d D:\libimobiledevice\DeveloperDiskImage.dmg Uploading D:\libimobiledevice\DeveloperDiskImage.dmg
ERROR: Unknown error occurred, can't mount.

D:\libimobiledevice>
~~~

cczhr, 2022-09-19:
Try turning on developer mode
[image]

mexmer, 2022-09-19:
did you enable developer mode on your device?
without developer mode enabled, developer mode doesn't work.

check following gist, but read discussion bellow, because code is not "perfect"
https://gist.github.com/nikias/262bd709c1651e0139eb9e3a2e2d33f4

xanadu-lee, 2022-09-20:
I mean, developer images can be uploaded. But can't mount.
So, there are no developer options on the device.

mexmer, 2022-09-20:
if you debugging ios 16.1 you need to use 16.1 image from Xcode beta
also yes, uploading developer image works, because it's done trough AFC service, but mounting requires DEVELOPER mode enabled

mexmer, 2022-09-20:
i have posted fixed gist from nikias
https://gist.github.com/mexmer/5c9dc1220a9d028d017d1eaa76fe8df2
compile it and use
```
idevicedevmodectl showmenu
```
to show developer options in settings on device (under privacy)
then togle developer mode on - it will restart device
then you need to confirm developer mode after device unlock
and that's it, you should see in settings it's on

xanadu-lee, 2022-09-20:
I think I get it.
I have successfully displayed the developer mode option on the device through the tutorial you sent.
Thank u.

closed 2022-09-20 by xanadu-lee

mexmer, 2022-09-20:
while i understand it's security measure, this new setting is just pain ...

xanadu-lee, 2022-09-20:
You're right, it's really painful.
Some functions that could have been used are now castrated.