libimobiledevice/libimobiledevice #1365: mac os 13 Can't install

liyunfei0618, 2022-09-19:
fatal: not in a git directory
Error: Command failed with exit 128: git
Warning: You are using macOS 13.
We do not provide support for this pre-release version.
You will encounter build failures with some formulae.
Please create pull requests instead of asking for help on Homebrew's GitHub,
Twitter or any other official channels. You are responsible for resolving
any issues you experience while you are running this
pre-release version.

Error: libimobiledevice: no bottle available!
You can try to install from source with:
  brew install --build-from-source libimobiledevice
Please note building from source is unsupported. You will encounter build
failures with some formulae. If you experience any issues please create pull
requests instead of asking for help on Homebrew's GitHub, Twitter or any other
official channels.

Schlaubischlump, 2022-09-21:
This is not related to libimobiledevice. This has to do with the fact that homebrew does not provide pre-build binaries for beta versions of macOS. 

Like the error message suggests you can try to install libimobiledevice using:

brew install --build-from-source libimobiledevice

Alternatively you can copy over the old binaries for macOS 12.0. They should work as well.

In every case this issue should be closed.

wangshouye, 2022-11-22:
try update brew to version 3.6.12 and update XCode to version 14.1