libimobiledevice/libimobiledevice #1370: idevicesetlocation setting speed along latitude and longitude

chmiiller, 2022-10-04:
Hi there, thanks for the awesome work you have already done!

I want to be able to simulate location with `lat` and `lon` but also be able to simulate speed, heading and altitude.
With `idevicesetlocation` we can simulate a position, but do you know if it's possible to extend it in order to support speed for example? I've tried a thing or two myself but honestly I couldn't understand how [@robbertvanginkel](https://github.com/robbertvanginkel) reversed this service when he exposed `idevicesetlocation` here https://github.com/libimobiledevice/libimobiledevice/pull/480 and if there's any possibility of setting a speed for example.

Thanks!

nikias, 2022-10-09:
Hi, unfortunately the service behind it doesn't accept anything else except latitude and longitude.

chmiiller, 2022-10-10:
thanks for the reply, that's sad to hear. Let's hope that Apple enables better location simulation one day in the future.

closed 2022-10-10 by chmiiller