libimobiledevice/libimobiledevice #1374: Package a new version

jiska2342, 2022-10-21:
Hi :smiley:  I just tried one of the new features of `libimobiledevice`, the `idevicebtlogger`, and it works like a charm on Linux.

Since there seem to be some substantial changes within the last two years, it would be awesome if you could package a new version so that these features are available without compiling the project.

Keep up the great work :fire:

xiaozhuai, 2023-02-03:
I vote this. We should release a new version for all `libmobiledevice/xxx` since there are many improvements since last two years.

mgr34, 2023-03-23:
I also vote for a 1.04 release. It seems maybe COVID impacted releases? Considering the last release date and the frequency in between other releases. 

I know many Linux distros will not update their packages until a new version is out.