libimobiledevice/libimobiledevice #1379: iOS 16.1.1 Could not perform backup protocol version exchange, error code -1

PaladinMorak, 2022-11-18:
Hello,

I am currently trying to make a backup of my iOS 16.1.1 device, but am getting the following error message
```
Started "com.apple.mobilebackup2" service on port 49423.
Could not perform backup protocol version exchange, error code -1
```
upon the recommendation of other closed issues I built libimobiledevice from scratch, but am still getting the same error. Any help would be appreciated, thanks in advance.

mexmer, 2022-11-21:
make sure your device is paired prio backup.

nikias, 2022-11-21:
It works with iOS 16.0, not sure if Apple changes something. Can you run `idevicebackup2` with `-d` to see some debug output?

mexmer, 2022-11-21:
@nikias just checked, and itunes backup now requires password confirmation to start actual backup. so that might be issue. while exchange version is still 400 (eg you send 400), it waits for you to enter password on device after exchange.

mexmer, 2022-11-21:
this is extract from communication log from our software, but looks normal
received
```
<?xml version="1.0" encoding="UTF-8"?>
<!DOCTYPE plist PUBLIC "-//Apple//DTD PLIST 1.0//EN"	"http://www.apple.com/DTDs/PropertyList-1.0.dtd">
<plist version="1.0">
<array>
<string>DLMessageVersionExchange</string>
<integer>400</integer>
<integer>0</integer>
</array>
</plist>
```
sent
```
<?xml version="1.0" encoding="UTF-8"?>
<!DOCTYPE plist PUBLIC "-//Apple//DTD PLIST 1.0//EN"	"http://www.apple.com/DTDs/PropertyList-1.0.dtd">
<plist version="1.0">
<array>
<string>DLMessageVersionExchange</string>
<string>DLVersionsOk</string>
<integer>400</integer>
</array>
</plist>
```
received
```
<?xml version="1.0" encoding="UTF-8"?>
<!DOCTYPE plist PUBLIC "-//Apple//DTD PLIST 1.0//EN"	"http://www.apple.com/DTDs/PropertyList-1.0.dtd">
<plist version="1.0">
<array>
<string>DLMessageVersionExchange</string>
<string>DLVersionsOk</string>
<integer>400</integer>
</array>
</plist>
```
sent
```
<?xml version="1.0" encoding="UTF-8"?>
<!DOCTYPE plist PUBLIC "-//Apple//DTD PLIST 1.0//EN"	"http://www.apple.com/DTDs/PropertyList-1.0.dtd">
<plist version="1.0">
<array>
<string>DLMessageProcessMessage</string>
<dict>
<key>LogLevel</key>
<integer>1</integer>
<key>MessageName</key>
<string>Hello</string>
<key>SupportedProtocolVersions</key>
<array>
<real>2</real>
<real>2.1000000000000000888</real>
</array>
</dict>
</array>
</plist>
```
received
```
<?xml version="1.0" encoding="UTF-8"?>
<!DOCTYPE plist PUBLIC "-//Apple//DTD PLIST 1.0//EN"	"http://www.apple.com/DTDs/PropertyList-1.0.dtd">
<plist version="1.0">
<array>
<string>DLMessageProcessMessage</string>
<dict>
<key>ErrorCode</key>
<integer>0</integer>
<key>MessageName</key>
<string>Response</string>
<key>ProtocolVersion</key>
<real>2.1000000000000000888</real>
</dict>
</array>
</plist>
```
sent
```
<?xml version="1.0" encoding="UTF-8"?>
<!DOCTYPE plist PUBLIC "-//Apple//DTD PLIST 1.0//EN"	"http://www.apple.com/DTDs/PropertyList-1.0.dtd">
<plist version="1.0">
<array>
<string>DLMessageProcessMessage</string>
<dict>
<key>MessageName</key>
<string>Backup</string>
<key>Options</key>
<dict>
<key>ForceFullBackup</key>
<true/>
</dict>
<key>TargetIdentifier</key>
<string>XXXXXXXXXXX</string>
</dict>
</array>
</plist>
````
after this, phone is waiting for passcode input, and will not continue communication.

PaladinMorak, 2022-11-21:
Yeah, I just ran it again and now it works. Might have been because I didn't restart the computer after I update usbmuxd and so the udev rules didn't load?

closed 2022-11-21 by PaladinMorak

mexmer, 2022-11-21:
> Yeah, I just ran it again and now it works. Might have been because I didn't restart the computer after I update usbmuxd and so the udev rules didn't load?

you don't need to restart computer, but you need to run ldconfig and restart usbmuxd

PaladinMorak, 2022-11-21:
Ah ok, good to know. Thanks for all the help.