libimobiledevice/libimobiledevice #1381: idevicebackup2 - IPhone not sending files??

RichardZimmerEschborn, 2022-11-23:
### Environment
- Debian Linux 11.5 stable up to date
-- providing idevicebackup2 Version 1.3.0
- Iphone 12 with IOS 16.1.1
- Both conntected via USB cable.

### Command executed
**$ idevicebackup2  backup   iphone**
Backup directory is "iphone"
Started "com.apple.mobilebackup2" service on port 50402.
Negotiated Protocol Version 2.1
Reading Info.plist from backup.
Starting backup...
Backup will be unencrypted.
Requesting backup from device...
Incremental backup mode.
ERROR: Could not receive from mobilebackup2 (-4)
Received 0 files from device.
Backup Aborted.

### Observed
After long time waiting and nothing happened I and my computer fell asleep.

First try created directory named like the UDID and in there a 1.4 MB file named Info.plist.

The file contains detailed information about the IPhone. Intersting this file states that it is an "iPhone13,2".

After a few trys at the end of this file it says:

```
        <key>iTunes Settings</key>
        <dict/>
        <key>iTunes Version</key>
        <string>12.10.10</string>

```
To me it looks like the communication with the phone is working but the phone is not sending data.

### Expected result
Backup of everything on IPhone

### Results
Only the mentioned directory and file Info.plist.
No other data.

mexmer, 2022-11-24:
as of iOS 16.1, when phone has passcode set, iTunes backup requires passcode to start as mentioned here #1380
flow is following, you start backup operation (run idevicebackup2)
on phone appears request for passcode, until you enter passcode on phone, phone will not send ANY files and eventually process will timeout, if you enter and confirm passcode, phone will start sending files and backup progresses as normaly.

RichardZimmerEschborn, 2022-11-25:
@mexmer 
Thanks for the hint.

I have seen that and entered the passcode.
I tried to verify if the Info.plist is not created without passcode, but it is.
From your mentioned issue #1380 I expect, that backup is working for you against IOS 16.x. Can you please verify, that it is still working against 16.1.1?

Is there any way to troubleshoot what is going on?

I have seen youtube videos that after a view seconds progressbars should show activity.

For me it never worked and it is always writing:
```
Requesting backup from device...
Full backup mode.
```
and than doing nothing.

mexmer, 2022-11-25:
yes, it works for me, i did backup of my ipad air 5 running ios 16.1.1
i did following
run idevicebackup2 backup
input passcode on ipad
wait for backup to finish

mind depending on device content, initial step (before it starts sending files), can take up even few minutes, i have friend that has iphone 13 pro 512GB with over 150 thousands of photos, it takes really long time, before phone starts sending anything
after backup command is sent, phone first requests manifest database from your computer (so it can check, what has been backed up in past), then it does enumerate content, to check what it needs to backup, and only after that it starts sending files.

of course, there is possibily, that backup process inside phone crashed during enumeration, or is deadlocked.

what you can try is to get syslog, after you run backup command and wait for let's say 1 minute. we might have some clue about what has happened from syslog.

sometimes if backup is "stuck" it might help to restart phone, also if you have computer with windows or macos installed, you can try to do local backup with itunes. since idevicebackup2 only simulates itunes backup.

RichardZimmerEschborn, 2022-11-26:
@mexmer 

What does it mean, that it may take a few minutes before backup starts? Also what means "it takes really long time"? I have 34.4 GB of storage used on the IPhone (mainly photographs and  4500 contacts) and waited for 30 minutes nothing happened.

In Iphone - settings - general - transfer IPhone/reset I completely reset all settings (to be sure also all network settings etc.) in the IPhone which did not help at all.

In syslog I found the following entry, which might give a hint:

`Nov 26 11:29:33 mobile gnome-shell[1594]: Unable to mount volume Dokumente auf iPhone: Gio.IOErrorEnum: Unbehandelter Fehler in Zugriffsrechten (-21)
`

For those not speaking German, the end means:
 Unhandeled Error in Permissions (-21).

I also found quite some more errors from gnome-shell, which I further investigate.

RichardZimmerEschborn, 2022-11-26:
@mexmer 

here some more messeges from syslog:

`
Nov 26 14:05:35 mobile kernel: [82381.201401] usb 1-2: new high-speed USB device number 18 using xhci_hcd
Nov 26 14:05:35 mobile kernel: [82381.353554] usb 1-2: New USB device found, idVendor=05ac, idProduct=12a8, bcdDevice=13.02
Nov 26 14:05:35 mobile kernel: [82381.353561] usb 1-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
Nov 26 14:05:35 mobile kernel: [82381.353565] usb 1-2: Product: iPhone
Nov 26 14:05:35 mobile kernel: [82381.353568] usb 1-2: Manufacturer: Apple Inc.
Nov 26 14:05:35 mobile kernel: [82381.353571] usb 1-2: SerialNumber: 00008101000B45491178001E
Nov 26 14:05:35 mobile mtp-probe: checking bus 1, device 18: "/sys/devices/pci0000:00/0000:00:14.0/usb1/1-2"
Nov 26 14:05:35 mobile mtp-probe: bus: 1, device: 18 was not an MTP device
Nov 26 14:05:35 mobile mtp-probe: checking bus 1, device 18: "/sys/devices/pci0000:00/0000:00:14.0/usb1/1-2"
Nov 26 14:05:35 mobile mtp-probe: bus: 1, device: 18 was not an MTP device
Nov 26 14:05:35 mobile kernel: [82381.462175] ipheth 1-2:4.2: Apple iPhone USB Ethernet device attached
Nov 26 14:05:35 mobile colord[1087]: CdMain: failed to emit DeviceAdded: failed to register object: An object is already exported for the interface org.freedesktop.ColorManager.Device at /org/freedesktop/ColorManager/devices/sysfs__null_
Nov 26 14:05:36 mobile colord[1087]: CdMain: failed to emit DeviceAdded: failed to register object: An object is already exported for the interface org.freedesktop.ColorManager.Device at /org/freedesktop/ColorManager/devices/sysfs__null_
Nov 26 14:05:36 mobile kernel: [82381.490294] ipheth 1-2:4.2 enxb28c75c3d925: renamed from eth0
Nov 26 14:05:36 mobile upowerd[1786]: treating change event as add on /sys/devices/pci0000:00/0000:00:14.0/usb1/1-2/power_supply/apple_mfi_fastcharge`

These messages are generated when plugging the IPhone to USB.

Especially the line with

`failed to register object`

makes me wonder. But what has backup to do with ColorManager?

mexmer, 2022-11-28:
@RichardZimmerEschborn i did mean syslog from phone, not from your linux machine,
 use idevicesyslog tool and search for backup related in it's output.

as for backup preparation phase, 30 minute is too long, also 34GB is not that much of data, should not take more than 2 minutes, on 256GB iphone (120GB of photos) i've seen backup preparation phase up to 4 minutes.

RichardZimmerEschborn, 2022-11-28:
@mexmer 

Thanks alot, this was the hint needed.

By scanning through the syslog of the IPhone I descovered a message complaining about not enough diskspace. So I checked local diskspace and really it is low.

Just starting idevicebackup2 on a different harddisk does the trick.

I am closing this issue and writing a new issue to request an error message if disk space is not suitable to hold the backup.

Thanks again ...

closed 2022-11-28 by RichardZimmerEschborn

mexmer, 2022-11-28:
> @mexmer
>
> Thanks alot, this was the hint needed.
>
> By scanning through the syslog of the IPhone I descovered a message complaining about not enough diskspace. So I checked local diskspace and really [...]

hmm, this is weird, because device should send this error back to computer, and idevicebackup2 should display error, like "mbdomain(code)/not enough disk space" or something similar, at least that's what it does for me, when there is not enough space on disk ... maybe there was change in 16.1, i will need to check for that later.

anyways, thanks for information, and i'm glad you have solved your problem and can make backup now.