libimobiledevice/libimobiledevice #1389: ERROR: build failed for libirecovery, check /Users/mac/limd-build/libirecovery_build.log for details.

amats211, 2022-12-13:
Trying to build limd-build-macos.sh by nikias but having that error when bulding irecovery. Below is the logs on build. Any help will be appreciated, thanks.

/Applications/Xcode.app/Contents/Developer/usr/bin/make  all-recursive
Making all in src
  CC       libirecovery_1_0_la-libirecovery.lo
  CCLD     libirecovery-1.0.la
Making all in include
make[2]: Nothing to be done for `all'.
Making all in tools
  CC       irecovery-irecovery.o
  CCLD     irecovery
ld: warning: ignoring file /usr/local/lib/libreadline.dylib, building for macOS-x86_64 but attempting to link with file built for unknown-unsupported file format ( 0x55 0x52 0x4C 0x3D 0x22 0x68 0x74 0x74 0x70 0x73 0x3A 0x2F 0x2F 0x69 0x75 0x6E )
Undefined symbols for architecture x86_64:
  "_add_history", referenced from:
      _append_command_to_history in irecovery-irecovery.o
  "_read_history", referenced from:
      _load_command_history in irecovery-irecovery.o
  "_readline", referenced from:
      _init_shell in irecovery-irecovery.o
  "_write_history", referenced from:
      _append_command_to_history in irecovery-irecovery.o
ld: symbol(s) not found for architecture x86_64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[2]: *** [irecovery] Error 1
make[1]: *** [all-recursive] Error 1
make: *** [all] Error 2

mexmer, 2022-12-13:
are you on M1/M2 mac?

amats211, 2022-12-13:
sorry forgot to mention, im on catalina..

mexmer, 2022-12-13:
looks to me either outdated or incompatibile libraries
try to update binutils and readline first with homebrew

or better, remove them, then install again.

then run autogen.sh in libirecovery folder and finally make all

SeaQuench, 2023-01-12:
i built this for Monterey without issue. good luck!