libimobiledevice/libimobiledevice #1390: Problems compiling.

ViewtifulCereza, 2022-12-15:
Hi, while comping I run into an issue and I'm trying to figure out what I could possibly missing here.
`CCLD     idevicebtlogger
/usr/bin/ld: ../src/.libs/libimobiledevice-1.0.so: undefined reference to opack_decode_to_plist
/usr/bin/ld: ../src/.libs/libimobiledevice-1.0.so: undefined reference to tlv_buf_append
/usr/bin/ld: ../src/.libs/libimobiledevice-1.0.so: undefined reference to tlv_data_get_uint
/usr/bin/ld: ../src/.libs/libimobiledevice-1.0.so: undefined reference to tlv_buf_new
/usr/bin/ld: ../src/.libs/libimobiledevice-1.0.so: undefined reference to tlv_buf_free
/usr/bin/ld: ../src/.libs/libimobiledevice-1.0.so: undefined reference to tlv_data_get_uint8
/usr/bin/ld: ../src/.libs/libimobiledevice-1.0.so: undefined reference to opack_encode_from_plist
/usr/bin/ld: ../src/.libs/libimobiledevice-1.0.so: undefined reference to tlv_data_copy_data
collect2: error: ld returned 1 exit status`

First thought was libplist, but I had successfully compiled and installed it.

mexmer, 2022-12-15:
i can't see any of mentioned functions in libimobile source, do you compile from this github, or from some patched 3rd party source?

ViewtifulCereza, 2022-12-15:
From this git repo.

mexmer, 2022-12-16:
still i can't find any reference in libimobile to any of these functions

tlv_data_copy_data
tlv_data_get_uint8t
tlv_buf_free
tlv_buf_new

and can't even find, what library contains this functions. it's something in your build enviroment, might have headers with same name. i would suggest compiling it on different computer, that will have installed only minimal set of stuff required for compiling libimobile.

ViewtifulCereza, 2022-12-16:
well it's here.
https://imgur.com/a/b31rtxr

mexmer, 2022-12-16:
i see, i don't use wireless pairing, that's why i don't see it, but seems you are missing libimobiledevice-glue, that's where tlv functions are.

first compile and install libimobiledevice-glue, then compile libimobiledevice, if you building on osx, it might be better to use nikias skript

ViewtifulCereza, 2022-12-16:
I sucessfully compiled and installed libimobiledevice-glue beforehand. I'm not specifically trying wireless pairing, just compiling libimobiledevice.
`whereis libimobiledevice-glue
libimobiledevice-glue: /usr/include/libimobiledevice-glue`

mexmer, 2022-12-16:
i think wireless pairing was probly enabled during configure phase, just because it detected it's possible.
i use libimobile on pc with linux, so this option doesn't enable for me ....

since you have libimobiledevice-glue headers in /usr/include, you have most likely installed libimobiledevice-glue from package
i suspect, that libimobiledeviceglue you have installed on your system might be outdated or not configured with wireless pairing support.

so either rebuild libmimobiledevice-glue from source, or (might be actually faster)
do 
```
make clean
./configre --disable-wireless-pairing

if you on macos, you can use nikias script, as i mentioned before
https://gist.github.com/nikias/84c79469a1d0f16ff95250f0d51858c3

ViewtifulCereza, 2022-12-16:
It's possible, yes, even though I did build it again from source. I'm on Arch Linux.  Unfortunate disabeling wireless did not solve the issue, still fails to the undefined references. For now I'll give up because, removing all the dependencies is too annoying, considering I just needed the new idevicedevmodectl enabling dev mode on iOS 16. Thanks anyways.

nikias, 2022-12-16:
This is for debian/ubuntu, but take a look: https://gist.github.com/nikias/e72902570cca679e752a7076a6b2127e

ViewtifulCereza, 2022-12-16:
@nikias thanks this script went through without any problems and I have access to idevicedevmodectl. No idea where I was stuck, but I guess I missed to build something on the way up.

closed 2022-12-16 by ViewtifulCereza