libimobiledevice/libimobiledevice #1391: The confusion while getting idevices color

Habibullah-pythoner, 2022-12-16:
When I use ideviceinfo on iPad, I get
(DeviceColor: black) exactly what I want

But when I try it on iPhone 6, I get the following:
(DeviceColor: #e1e4e3)

and when I try iPhone 11, I get:
(DeviceColor: 1)

Is there a way to get only one type of output (the color name would be nice, eg. Black, Midnight Green...), instead of all these different and confusing results
even a way to convert these outputs to the brand's English color name would be nice too (as mentioned eg. Black, Silver, Gold....)

mexmer, 2022-12-16:
these values are supplied by device, so whatever apple written there is there.

Habibullah-pythoner, 2022-12-17:
so any source to know what these values mean?
so I can convert them manually

mexmer, 2022-12-17:
to be honest, i have no idea, you would need to look around internet.
ideviceinfo (you could also run ```ideviceinfo -x``` to get xml output, as was received from device) only returns what device sent, nothing less, nothing more, it doesn't interpret values

you can referer to this thread from past, where they were asking same question
https://github.com/libimobiledevice/libimobiledevice/issues/818

Habibullah-pythoner, 2022-12-17:
understandable, so let's think outside the box
is there a way to decode the color from, I don't know, maybe serial number, imei or maybe model number?

nikias, 2022-12-17:
Check https://www.theiphonewiki.com/wiki/Models - every color has its own model number.

Habibullah-pythoner, 2022-12-17:
Right, I did check on that site, I even wrote all of the phones section model numbers, but after some test, I realized that the site lacks a lot of model numbers.
Not much accurate data

closed 2023-01-22 by Habibullah-pythoner