libimobiledevice/libimobiledevice #1395: Lockdownd, couldn't connect bcs of (SSL ERROR (-5)) [IDEVICEINFO]

Habibullah-pythoner, 2022-12-24:
/ideviceinfo works fine with iphones that are activated, but doesnt work will with iphones that are none active (its like luck, 50% of times it works), it returns the error below.
Couldn't connect to lockdownd device, SSL error (-5)

Habibullah-pythoner, 2022-12-24:
here is the exact image, please let me know if further information or details is needed.
[https://imgur.com/o05UFiQ](url)

Thanks

shubhamkd, 2022-12-31:
in order to re-generate this issue, will have to buy new iPhone xD

mexmer, 2022-12-31:
it's limitation of newer ioses, since setting up SSL connection requires that you confirm trust on device, which dialog doesn't pop on device, that is in setup process. so yes, you need to finish setup and activate your device first.

nikias, 2023-01-01:
No this is not true. A device in setup mode is paired automatically without trust. Usually an old pairing record would result in such an SSL error. Try idevicepair unpair and then reconnect the device and run ideviceinfo again and it will work.

mexmer, 2023-01-01:
@nikias maybe it depends on which phase you are in setup, i will recheck it later. i know, right after fatory reset, you can restore your device and it will autopair, but when you go to step with activation lock, until you do it, it will fail to pair.

maybe it's also related to activation lock, not sure.

still you are right, that `idevicepair unpair` and `idevicepair pair` should be first thing to do, in case of ssl error, i forgot about that.

tarikki, 2023-04-29:
So does the old pairing record mean that the computer has seen the iPhone before, has paired with it but is somehow unable to do that again? 

Btw Could anyone point me to some good resources on the subject of pairing records? I tried searching the webs but information was surprisingly scarce.