libimobiledevice/libimobiledevice #1397: idevice_connection_disable_bypass_ssl(): SSL mode disabled

AldeabAhmed, 2022-12-31:
Failed to arm Developer Mode (1)
==============================
Last login: Sat Dec 31 13:25:47 on ttys002
(base) ahmedaldeab@Aldeab-MacBook-Pro ~ % idevicedevmodectl enable -d
13:48:50 lockdown.c:636 lockdownd_client_new(): device udid: *************************
13:48:50 lockdown.c:365 lockdownd_query_type(): called
13:48:50 property_list_service.c:132 internal_plist_send(): sending 290 bytes
13:48:50 service.c:144 service_send(): sending 4 bytes
13:48:50 idevice.c:692 idevice_connection_send(): internal_connection_send 4, sent 4
13:48:50 service.c:144 service_send(): sending 290 bytes
13:48:50 idevice.c:692 idevice_connection_send(): internal_connection_send 290, sent 290
13:48:50 property_list_service.c:137 internal_plist_send(): sent 290 bytes
13:48:50 property_list_service.c:138 internal_plist_send(): printing 290 bytes plist:
<?xml version="1.0" encoding="UTF-8"?>
<!DOCTYPE plist PUBLIC "-//Apple//DTD PLIST 1.0//EN" "http://www.apple.com/DTDs/PropertyList-1.0.dtd">
<plist version="1.0">
<dict>
	<key>Label</key>
	<string>idevicedevmodectl</string>
	<key>Request</key>
	<string>QueryType</string>
</dict>
</plist>
13:48:50 property_list_service.c:205 internal_plist_receive_timeout(): initial read=4
13:48:50 property_list_service.c:211 internal_plist_receive_timeout(): 297 bytes following
13:48:50 property_list_service.c:224 internal_plist_receive_timeout(): received 297 bytes
13:48:50 property_list_service.c:253 internal_plist_receive_timeout(): printing 297 bytes plist:
<?xml version="1.0" encoding="UTF-8"?>
<!DOCTYPE plist PUBLIC "-//Apple//DTD PLIST 1.0//EN" "http://www.apple.com/DTDs/PropertyList-1.0.dtd">
<plist version="1.0">
<dict>
	<key>Request</key>
	<string>QueryType</string>
	<key>Type</key>
	<string>com.apple.mobile.lockdown</string>
</dict>
</plist>
13:48:50 lockdown.c:381 lockdownd_query_type(): success with type com.apple.mobile.lockdown
13:48:50 property_list_service.c:132 internal_plist_send(): sending 338 bytes
13:48:50 service.c:144 service_send(): sending 4 bytes
13:48:50 idevice.c:692 idevice_connection_send(): internal_connection_send 4, sent 4
13:48:50 service.c:144 service_send(): sending 338 bytes
13:48:50 idevice.c:692 idevice_connection_send(): internal_connection_send 338, sent 338
13:48:50 property_list_service.c:137 internal_plist_send(): sent 338 bytes
13:48:50 property_list_service.c:138 internal_plist_send(): printing 338 bytes plist:
<?xml version="1.0" encoding="UTF-8"?>
<!DOCTYPE plist PUBLIC "-//Apple//DTD PLIST 1.0//EN" "http://www.apple.com/DTDs/PropertyList-1.0.dtd">
<plist version="1.0">
<dict>
	<key>Label</key>
	<string>idevicedevmodectl</string>
	<key>Key</key>
	<string>ProductVersion</string>
	<key>Request</key>
	<string>GetValue</string>
</dict>
</plist>
13:48:50 property_list_service.c:205 internal_plist_receive_timeout(): initial read=4
13:48:50 property_list_service.c:211 internal_plist_receive_timeout(): 325 bytes following
13:48:50 property_list_service.c:224 internal_plist_receive_timeout(): received 325 bytes
13:48:50 property_list_service.c:253 internal_plist_receive_timeout(): printing 325 bytes plist:
<?xml version="1.0" encoding="UTF-8"?>
<!DOCTYPE plist PUBLIC "-//Apple//DTD PLIST 1.0//EN" "http://www.apple.com/DTDs/PropertyList-1.0.dtd">
<plist version="1.0">
<dict>
	<key>Key</key>
	<string>ProductVersion</string>
	<key>Request</key>
	<string>GetValue</string>
	<key>Value</key>
[... 456 lines cut ...]
13:48:50 property_list_service.c:211 internal_plist_receive_timeout(): 238 bytes following
13:48:50 idevice.c:944 internal_ssl_read(): pre-read length = 5 bytes
13:48:50 idevice.c:961 internal_ssl_read(): read 5 bytes
13:48:50 idevice.c:970 internal_ssl_read(): post-read received 5 bytes
13:48:50 idevice.c:944 internal_ssl_read(): pre-read length = 262 bytes
13:48:50 idevice.c:961 internal_ssl_read(): read 262 bytes
13:48:50 idevice.c:970 internal_ssl_read(): post-read received 262 bytes
13:48:50 idevice.c:813 idevice_connection_receive_timeout(): SSL_read 238, received 238
13:48:50 property_list_service.c:224 internal_plist_receive_timeout(): received 238 bytes
13:48:50 property_list_service.c:253 internal_plist_receive_timeout(): printing 238 bytes plist:
<?xml version="1.0" encoding="UTF-8"?>
<!DOCTYPE plist PUBLIC "-//Apple//DTD PLIST 1.0//EN" "http://www.apple.com/DTDs/PropertyList-1.0.dtd">
<plist version="1.0">
<dict>
	<key>Request</key>
	<string>StopSession</string>
</dict>
</plist>
13:48:50 lockdown.c:262 lockdownd_stop_session(): success
13:48:50 idevice.c:982 internal_ssl_write(): pre-send length = 31 bytes
13:48:50 idevice.c:988 internal_ssl_write(): post-send sent 31 bytes
13:48:50 idevice.c:1443 idevice_connection_disable_bypass_ssl(): Skipping bidirectional SSL shutdown. SSL error code: 5
13:48:50 idevice.c:1460 idevice_connection_disable_bypass_ssl(): SSL mode disabled
***********************: Failed to arm Developer Mode (1)
(base) ahmedaldeab@Aldeab-MacBook-Pro ~ %

mexmer, 2022-12-31:
if you have passcode set on your device, `idevicedevmodectl enable` and 'idevicedevmodectl arm` doesn't work, you need to do it manually on your device. 
first you enable developer mode settings with `idevicedevmodectl reveal`, go to security->development and enable it.  after phone restart you confirm it manually.

aslo developer mode is required on ios 16, if your device running older version of ios, you don't need to do this.

AldeabAhmed, 2022-12-31:
Currently, the device is full restored and it’s  unactivated and does not have any passcode or anything else, because it has a defective baseband
So, that is why I want to enable Dev mode :/

nikias, 2022-12-31:
The error is `InvalidService` and since you said the device is not activated that must be the reason the amfi service is not available.

AldeabAhmed, 2022-12-31:
from what I understood from you, the tool cannot work while the device is not activated! or did I miss something?

nikias, 2022-12-31:
Correct. A bunch of things don't work when the device is not activated.