libimobiledevice/libimobiledevice #1398: Script to build libi is having a issue - Undefined symbols for architecture arm64

KevHerna, 2023-01-04:
I tried to use the Script you mention to correctly build Libi but I'm having this problem:

ERROR: build failed for libimobiledevice

 Undefined symbols for architecture arm64:
  "_EVP_chacha20_poly1305", referenced from:
      _chacha20_poly1305_encrypt_96 in lockdown-cu.o
      _chacha20_poly1305_decrypt_96 in lockdown-cu.o
  "_RSA_meth_get_bn_mod_exp", referenced from:
      _BigIntegerUseEngine in libsrp6a-sha512.a(t_math.o)
  "_SSL_CTX_set_security_level", referenced from:
      _idevice_connection_enable_ssl in idevice.o
  "_X509_set1_notAfter", referenced from:
      _pair_record_generate_keys_and_certs in libinternalcommon.a(userpref.o)
  "_X509_set1_notBefore", referenced from:
      _pair_record_generate_keys_and_certs in libinternalcommon.a(userpref.o)
ld: symbol(s) not found for architecture arm64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[2]: *** [libimobiledevice-1.0.la] Error 1
make[1]: *** [all-recursive] Error 1
make: *** [all] Error 2
       

On macOS, you can easily build everything with my script:
```bash
mkdir -p limd-build
cd limd-build
curl -o ./limd-build-macos.sh -Ls https://git.io/JEhrV 
bash ./limd-build-macos.sh
```

_Originally posted by @nikias in https://github.com/libimobiledevice/libimobiledevice/issues/1261#issuecomment-1002986891_

KevHerna, 2023-01-04:
@nikias could you please help with this?

nikias, 2023-01-04:
Are you building on M1 or intel mac?

nikias, 2023-01-04:
The link you posted above is pointing to an older build script, are you positive you used the latest version?

KevHerna, 2023-01-04:
> Are you building on M1 or intel mac?

I'm using Intel currently 
MacBook Pro (16-inch, 2019)
2,6 GHz 6-Core Intel Core i7

KevHerna, 2023-01-04:
> The link you posted above is pointing to an older build script, are you positive you used the latest version?

Yes, I'm using this one: https://gist.githubusercontent.com/nikias/84c79469a1d0f16ff95250f0d51858c3/raw/43a3b8be413785085bc6653844cef57a73f9a12d/limd-build-macos.sh

nikias, 2023-01-04:
That's 1.0.7 but latest script is v1.0.10, so please check again that you have the latest:
https://gist.githubusercontent.com/nikias/84c79469a1d0f16ff95250f0d51858c3/raw/9ecb66e5d955982229d3b5cec63512b8bce59644/limd-build-macos.sh

KevHerna, 2023-01-04:
> That's 1.0.7 but latest script is v1.0.10, so please check again that you have the latest: https://gist.githubusercontent.com/nikias/84c79469a1d0f16ff95250f0d51858c3/raw/9ecb66e5d955982229d3b5cec63512b8bce59644/limd-build-macos.sh

I used that one you comment and I get the same problem:

ERROR: build failed for libimobiledevice, check /Users/user/limd-build/libimobiledevice_build.log for details.

[libimobiledevice_build.log](https://github.com/libimobiledevice/libimobiledevice/files/10346558/libimobiledevice_build.log)

KevHerna, 2023-01-05:
The reason I'm trying to compile with this is because I'm having the same problem as in this previously reported issue https://github.com/libimobiledevice/libimobiledevice/issues/1261#issuecomment-1002986891

Could not connect to screenshot! @nikias with installing the dependencies using Brew

nikias, 2023-01-05:
What's your macOS version, just for reference?

KevHerna, 2023-01-05:
> What's your macOS version, just for reference?

macOS Big Sur Version 11.7.2 @nikias

nikias, 2023-01-05:
It seems like it uses wrong OpenSSL headers, do you have any OpenSSL headers in `/usr/local/include/openssl` (or even `/usr/include/openssl` ) ?

KevHerna, 2023-01-05:
> It seems like it uses wrong OpenSSL headers, do you have any OpenSSL headers in `/usr/local/include/openssl` (or even `/usr/include/openssl` ) ?

On /usr/local/include 

[image]

The other path (/usr/include/openssl) doesn't exist

nikias, 2023-01-06:
Can you run this command: `find / -name opensslv.h`

KevHerna, 2023-01-06:
> Can you run this command: `find / -name opensslv.h`

**These are all the occurrences it found @nikias:**

/usr/local/Cellar/openssl@3/3.0.7/include/openssl/opensslv.h
/usr/local/Cellar/openssl@1.1/1.1.1s/include/openssl/opensslv.h

/System/Volumes/Data/usr/local/Cellar/openssl@3/3.0.7/include/openssl/opensslv.h
/System/Volumes/Data/usr/local/Cellar/openssl@1.1/1.1.1s/include/openssl/opensslv.h

/Users/user/limd-build/deps/libressl-2.2.7/include/openssl/opensslv.h

/dev/fd/6/usr/local/Cellar/openssl@3/3.0.7/include/openssl/opensslv.h
/dev/fd/6/usr/local/Cellar/openssl@1.1/1.1.1s/include/openssl/opensslv.h

/dev/fd/6/System/Volumes/Data/usr/local/Cellar/openssl@3/3.0.7/include/openssl/opensslv.h
/dev/fd/6/System/Volumes/Data/usr/local/Cellar/openssl@1.1/1.1.1s/include/openssl/opensslv.h

/dev/fd/6/System/Volumes/Data/Users/user/limd-build/deps/libressl-2.2.7/include/openssl/opensslv.h

/dev/fd/6/Users/user/limd-build/deps/libressl-2.2.7/include/openssl/opensslv.h

nikias, 2023-01-06:
Try to rename the paths:
```
sudo mv /usr/local/Cellar/openssl@3/3.0.7/include/openssl /usr/local/Cellar/openssl@3/3.0.7/include/_openssl
sudo mv /usr/local/Cellar/openssl@1.1/1.1.1s/include/openssl /usr/local/Cellar/openssl@1.1/1.1.1s/include/_openssl
```
Then run the build script again?
Afterwards you can restore it like this:

```
sudo mv /usr/local/Cellar/openssl@3/3.0.7/include/_openssl /usr/local/Cellar/openssl@3/3.0.7/include/openssl
sudo mv /usr/local/Cellar/openssl@1.1/1.1.1s/include/_openssl /usr/local/Cellar/openssl@1.1/1.1.1s/include/openssl
```

KevHerna, 2023-01-06:
It works @nikias :D

nikias, 2023-01-06:
Ok thanks for confirming. Could you do me a favor, and with the non-renamed (original) directories run the script again until it fails, then send me the `libimobiledevice/configure` (not `configure.ac`) script? You don't have to build again after that since you already succeeded installing a working build.

KevHerna, 2023-01-06:
> Ok thanks for confirming. Could you do me a favor, and with the non-renamed (original) directories run the script again until it fails, then send me the `libimobiledevice/configure` (not [...]

Sure, give me a moment.

KevHerna, 2023-01-06:
I have run the Script again with the folders as they were originally and it hasn't failed again, that's weird. @nikias

mexmer, 2023-01-09:
delete **configure.cache** and **configure.status** file, then run configure again

nikias, 2023-01-18:
@KevHerna feel free to post an update here, but I'll close it since the issue is resolved.

closed 2023-01-18 by nikias

KevHerna, 2023-05-12:
Thank you very much for your help @nikias 
Currently I am having another problem and it is that in mac M1 when I try to install ios-webkit-debug-proxy it reinstalls libimobile and discards the compiled with your file, therefore Screenshot stops working again.

KevHerna, 2023-05-18:
[libimobiledevice_build.log](https://github.com/libimobiledevice/libimobiledevice/files/11508557/libimobiledevice_build.log)