libimobiledevice/libimobiledevice #1399: error: Could not receive from mobilebackup2 (-4)

hcropps, 2023-01-05:
I'm getting the backup with no errors. I am trying to restore this backup without making any changes. I am getting "Could not receive from mobilebackup2 (-4)" error while restoring. When I look at the logs with syslog, it gives the following error. Why am I getting this error?

syslog_callback line Jan  5 23:41:46 iPhone kernel(AppleUSBDeviceMux)[0] <Notice>: detected duplicate packet. Expected 14992 received 15361
syslog_callback line Jan  5 23:41:46 iPhone kernel(IOUSBDeviceFamily)[0] <Notice>: 005657.519590 AppleNubSynopsysOTG3Device@7: IOUSBDeviceController::gated_setProperties: unknown command DumpDebug

nikias, 2023-01-06:
Looks like bad communication to me, maybe try with a different cable? Also, are you using Linux or a different platform?

hcropps, 2023-01-06:
@nikias  thanks. i am using android. I tried with a different cable, the result did not change. I completely reset my iphone. still same result. Also, I noticed that when I log the libimobile side, I get these two errors. Actually, I was able to do both backup and restore before. Now when I try, I can make a backup, but I can't restore it.

Errors I get:
DEVICE_LINK_SERVICE_E_MUX_ERROR
DEVICE_LINK_SERVICE_E_SSL_ERROR