libimobiledevice/libimobiledevice #1400: libssl.3.dylib (mach-o file, but is an incompatible architecture (have (arm64), need (x86_64)))

VijayYD, 2023-01-09:
When I tried to compile libimobiledevice on montary OS with M1 chip, getting ssl issue.But the same binary is working fine on Ventura with M1. 
Can you please help to resolve openssl issue for montary OS? It seems like binary is picking openssl1 where as on montary its finding openssl3. 
Did uninstall openssl and installed open ssl3 and tried force copying openssl3 into above path, still its giving same issue.

nikias, 2023-01-09:
I would suggest to choose a different installation method. I have prepared this script to build on macOS. It can fail though, if you have other openssl package(s) installed e.g. through homebrew. Try this:
```bash
mkdir -p limd-build;
cd limd-build;
curl -o ./limd-build-macos.sh -L is.gd/LlCh2k ;
bash ./limd-build-macos.sh
```

SeaQuench, 2023-01-12:
i am using Monterrey. i have both openssl@1 and openssl@3 installed.
```
For compilers to find openssl@3 you may need to set:
  export LDFLAGS="-L/usr/local/opt/openssl@3/lib"
  export CPPFLAGS="-I/usr/local/opt/openssl@3/include"
 
For pkg-config to find openssl@3 you may need to set:
  export PKG_CONFIG_PATH="/usr/local/opt/openssl@3/lib/pkgconfig"
```

nikias, 2023-01-13:
@VijayYD If you just want to configure and compile libimobiledevice you can also just set the architecture via CFLAGS:
```bash
CFLAGS="-arch arm64 -isysroot `xcrun --sdk macosx --show-sdk-path` ./autogen.sh
```
It seems like it finds your openssl libs but wants to build for intel platform, so the above might work.