libimobiledevice/libimobiledevice #1403: browsing and viewing photos in backup

dan-da, 2023-01-17:
*My use case:*

I am trying to incrementally backup photos from my wife's iphone onto a dedicated SSD drive via a linux laptop.  I wish to view the photos in a photo album manager, such as shotwell.  I am using an old xubuntu 16.04.  So basically the desired flow is:
1. Wife fills up phone and complains.
2. I run incremental backup to SSD
3. wife deletes all or most photos/videos on phone.
4. wife and I can view the photos/vids whenever we like on the laptop via a photo album manager.
5. goto 1.

*My experience:*
When I first ran idevicebackup2 I was hoping/expecting it would create a big DCIM type tree with all camera photos in one directory or a tree, but at least with .png or .jpeg extension, and separate from other files on the phone.

Instead, I find that all files are sort of intermixed in a bucket type folder tree and worse the image files have no file extension.  So when I point shotwell at this 102Gb directory, it processes for a long while and then reports it didn't find any photos or videos.  However if I run a filemanager eg thunar, it shows thumbnails of the images, so clearly they are recognizable by actually peeking inside each file.

I also tried digikam (from ubuntu 16.04), and it was able to recognize the images, but it only seems to support copying them into its own managed folder, whereas shotwell provides the option to copy or use existing files in place.  

I do not see any flags for idevicebackup2 to obtain the desired layout.  Nor to preserve timestamps of the photo files.  I notice that all files have create date of the moment of backup, not when the photo was taken.

It appears to me that idevicebackup2 is focuses somewhat narrowly on the backup/restore case (which is fine and useful) but perhaps it is not really contemplating that users will want to actually view the backed up photos.    ie, not contemplating the above flow.

I am curious to hear how others are using this, and/or would achieve the above flow.

thanks!!

ps: I notice there is an `idevicebackup2 list` command that appears to generate a list of the original pathnames.   So maybe it is possible to generate a shadow tree with symlinks that point to the corresponding backup file.  Does such a tool already exist perhaps?

mexmer, 2023-01-17:
don't use idevicebackup2 for backing up photos, idevicebackup does device backup.
if you want to copy photos out of your device, use filemanagemer, mount your device with ifuse, go to DCIM folder, and there are your photos.

dan-da, 2023-01-18:
thx @mexmer.   interesting, I was thinking that idevicebackup2 is essentially doing an incremental rsync type operation with a no-delete option.  essentially append only, time-machine-esque.   apparently not.

The ifuse suggestion is a good one.  then I can use an existing photo rsync script I have for android.  I would've started with that if I was aware of it, but the suggestions I found on a couple sites were to use idevicebackup2.

Perhaps it would be worthwhile to highlight this use case in the README file / docs, because I think it is super common.

nigoroll, 2023-07-22:
I use https://github.com/inflex/ideviceunback to extract the backup, which makes photos from the backup available in `Media/DCIM`. With the `-l` option this does not even require additional disk space, because hard links are used. But the "unback" directory (whatever specified using the `-o`) option needs to be deleted before the next backup, or you might end up with zero sized backup files.