libimobiledevice/libimobiledevice #1404: idevicebackup2 global progress indicator

dan-da, 2023-01-17:
Hi, I only see progress per file, but have no idea how long until the whole backup process finishes.

Is it possible to add some type of global progress indicator?

mexmer, 2023-01-17:
unfortunately not, you don't know exact size of backup, also backup has multiple stages (preparation, checks, purging removed files, making snapshot, moving snapshot to main backup folder, finalizing), while device report some kind of progress, it offten stays stuck at same value, sometimes even backtracks, you can't even calculate estimation based on trasnfer speed. (it's also reason why itunes on windows, do not show real backup progres and just cycle bar). most you can do is observer messages regarding stages ... but that's about it. with information that can be retrieved from device, it's not even worth doing educated guess.
while single file transfer progress is simple, because you know how much bytes will be trasfered.

nikias, 2023-01-17:
If you look closeley there are two progress bars, one for the individual file and a global progress one. I would have to do some math to estimate the estimated duration, but it would be based on the size and transferred data. Maybe I add this at some point.

mexmer, 2023-01-17:
@nikias i think somewhere in the past, i already posted, how idevicebackup sends progress, but you might find the "global" one behaves rather improperly. in older versions of iOS (prior ios 10), you could use notification observer, or you could add verbose flag to backupoptions (in later case, there was one additional value, that was like global backup percentage ... this is still available but it's horribly broken since ios 10, for some stages, it shows stage progres ... like when moving files from snaphost to main folder, then it reset back to global, so if you use it, it jumps all over the place)