libimobiledevice/libimobiledevice #1407: idevicesetlocation not working on iOS 16.1 version of iPhone14 series devices

tyrocoding, 2023-02-02:
```
~ admin$ idevicesetlocation -u 00008120-000E5D9111B8C01E -- 12.46 88.43
~ admin$ idevicesetlocation -u 00008120-000E5D9111B8C01E -- 12.4615242542 88.433232323
~ admin$ idevicesetlocation -u 00008120-000E5D9111B8C01E -- 12.4615242542 88.4332323231
```

On running idevicesetlocation gps location of the device is not changing on any of the location based apps. Facing this only with iPhone14 series models
There is no error on running this command. But location does not get updated with lat long passed

cameroncc, 2023-03-04:
The same thing happens to me on my iPhone 14. I can run some other commands, like `ideviceimagemounter` and `ideviceinfo`, but `idevicesetlocation` does not work.

cameroncc, 2023-03-22:
@nikias Have you seen this bug with any iPhone 14 series devices you may have? Is there anything I can do to help debug this issue?

Pranav-19, 2023-03-23:
Facing the exact same problem, `idevicesetlocation` not working on a few iPhone 14s on iOS 16.1

nikias, 2023-03-23:
Well, does idevicesyslog show anything?

Schlaubischlump, 2023-03-23:
The problem was already reported on one of my projects, where I use libimobiledevice to fake the location of an iOS device. 

To my knowledge:
1. It's an apple bug that only exists on iPhone 14 and 14 p
Pro on 16.1 (also happens when you use Xcode's location spoofing feature). 
2. It is fixed in all later versions of iOS 16. 
3. The bug does not occur if you put your device in airplane mode before you fake the location. 

Feel free to correct me if you experience a different behavior.