libimobiledevice/libimobiledevice #1409: idevicedebug run incorrectly eats any parameters after the bundle ID if they look like switches

zaphodikus, 2023-02-03:
The parsing of the parameters in idevicedebug seems to thing anything that looks like a switch after the run command is not inteded for the target app.
idevicedebug -u <someid> run mydomain.myapp.appname Arg0 Arg1

works just fine, but if your apps args are switches, the parsign parses them even if that appear after the 'run' command

idevicedebug -u <someid> run mydomain.myapp.appname -someflag value 
ends up not passing -someflag value to the target app at all because of the leading '-', quoting does not help, the tool eats the parameter and only passes 'value' through in this case.

Tool should stop reading args once it has the bundle id I think.

nikias, 2023-02-03:
That's because it uses getopt. Pass a `--` as argument and it will stop processing anything starting with a `-`.

zaphodikus, 2023-02-03:
Aha, that helps. Thank you. It must have been a long time since I saw anyone use this trick, it's not common knowledge, but as far I'm concerned that fixes the issue for me as long as someone else can see this answer.

Happy to close, adding '--' after the bundle ID solves it.

closed 2023-02-10 by zaphodikus