libimobiledevice/libimobiledevice #1411: idevicebackup2, can't make it work on Monterey and ipados-16.3

Jakketet, 2023-02-14:
installed libimobiledevice with Brew, first removed all old versions and installed clean. Tried ideviceinfo, idevicedate, idevice_id and these worked, then tried idevicebackup2 but failed with following error:

_% idevicebackup2 backup /Users/xx/Documents/ipad6
 Backup directory is "/Users/xx/Documents/ipad6"
 Started "com.apple.mobilebackup2" service on port 50267.
 Could not perform backup protocol version exchange, error code -1_

Any clues? All comments on this error seems to point to a version incompatibility but I think I got the last version...

outsourcestudio, 2023-02-24:
@Jakketet 
you need to compile actual version from source.
hombrew has old version that doesn't support current OS
see
https://github.com/libimobiledevice/libimobiledevice/issues/1299

Jakketet, 2023-02-24:
ok! thanks, did not notice #1299... my fault.
J.