libimobiledevice/libimobiledevice #1413: ERROR: Could not receive from mobilebackup2 (-4)

pogopaule, 2023-02-18:
This used to work. Did an iOS update break idevicebackup2? I am currently on iOS 16.1.1.

```
❯ idevicebackup2 -d backup /tmp/ibackup
Backup directory is "/tmp/ibackup"
Started "com.apple.mobilebackup2" service on port 49330.
Negotiated Protocol Version 2.1
Starting backup...
Backup will be encrypted.
Requesting backup from device...
Full backup mode.
ERROR: Could not receive from mobilebackup2 (-4)
Received 0 files from device.
Backup Aborted.
```

outsourcestudio, 2023-02-18:
@pogopaule recently I worked with Xcode version and got the same error.
https://github.com/libimobiledevice/libimobiledevice/issues/1412
I solved it in when increase default timeout from 30 sec to 90 sec. I saw device logs to understood device spends 62 sec to start backup.

pogopaule, 2023-02-18:
@outsourcestudio Thank you for your answer! That sounds promising. I can't figure out how to set a timeout...

outsourcestudio, 2023-02-18:
@pogopaule just in source code and build library. I do not know another way right now.

pogopaule, 2023-02-18:
@outsourcestudio do you remember if it was this value? https://github.com/libimobiledevice/libimobiledevice/blob/9a5e78fe56267d6c721c92055c3ae9702bd25871/src/property_list_service.c#L272

outsourcestudio, 2023-02-21:
Hello @pogopaule ! Sorry just saw your mention.
I changed replaced 30000 by 90000 twice

1)
LIBIMOBILEDEVICE_API service_error_t service_receive(service_client_t client, char* data, uint32_t size, uint32_t *received)
{
	return service_receive_with_timeout(client, data, size, received, 90000);
}
and here
2) LIBIMOBILEDEVICE_API property_list_service_error_t property_list_service_receive_plist(property_list_service_client_t client, plist_t *plist)
{
	return internal_plist_receive_timeout(client, plist, 90000);
}

mexmer, 2023-02-28:
you might need up to 2 minute timeout on iphone with ton of photos (on this one they used 190GB out of 512GB phone storage)

jshcmpbll, 2023-05-09:
@outsourcestudio This fixed it for me, thank you!

arksunix, 2023-05-29:
> Interesting sidenote: `-4` seems to be an SSL error, not a timeout error which is `-5` see:
> 
> https://github.com/libimobiledevice/libimobiledevice/blob/15f8652126664e3a4b980e5d1c039b9053ce8566/include/libimobiledevice/mobilebackup.h#L42

I fixed error by building with OpenSSL instead GnuTLS as backend

moranbw, 2024-04-23:
> I fixed error by building with OpenSSL instead GnuTLS as backend

How do you do this? Is this something that is going to be adopted by the project?

OctopusET, 2024-11-09:
On arch, I tried changing GnuTLS to OpenSSL. It didn't work. I had to change the timeout to 90000. Then it worked!

mexmer, 2024-11-11:
Timeout usually needs to be increased if you have lot of photos in your device, or lot of applicaiton data. in our app we increased timeout to 15 minutes, because device was 1TB iphone with over 100 thousand photos (not icloud synced(, and it really did took like 12 minutes, before we received reply from phone. probly adding optional timeout to commandline parameters of idevicebackup2 tool might be good idea.