libimobiledevice/libimobiledevice #1428: E: Unable to locate package libimobiledevice-glue-dev

Hayden959, 2023-03-28:
I can install every package besides this one it errors out what can i do?

E: Unable to locate package libimobiledevice-glue-dev

mexmer, 2023-03-29:
if your distribution doesn't contain this package, you need to compile from source code

Hayden959, 2023-03-29:
worked thanks

flyfishMT, 2023-04-07:
For clarity, you mean compile and install this repo from the master branch? https://github.com/libimobiledevice/libimobiledevice-glue

mexmer, 2023-04-07:
yes

daguza, 2023-12-13:
> worked thanks

@Hayden959 please be kind, and give feedback to know how did you succeed on fix it... On
https://github.com/libimobiledevice/libimobiledevice-glue

section  _installation / getting started_ I followed instructions, but my system is still with message error:
`E: Unable to locate package libimobiledevice-glue-dev`

Thanks in advance