libimobiledevice/libimobiledevice #1429: ERROR: Could not connect to lockdown: Mux error (-8) on Windows

didix21, 2023-03-30:
### Environment

**Libimobiledevice**
- reference: `c797c2af6f4e2e8ae93b68f1c41abab1579563db`

**Machine**
- OS: Windows 10 - 10.0.19045
- Apple Mobile Device Support: 14.5.0.7
- Bonjour: 3.1.0.1

**Device**
- Device type: iPhone 8
- OS: iOS 16.3.1

### Description

I've enabled 'Show this iPhone when on WiFi' from iTunes. It seems that when I run `idevice_id.exe`, it returns the device's UDID from the network. However, when I run `ideviceinfo.exe -n -u <udid>` or `idevicebackup2.exe -n -u <udid> backup <path>`, I receive the following error: 'ERROR: Could not connect to lockdown: Mux error (-8)'. Maybe it's related to the following issue: https://github.com/libimobiledevice/libimobiledevice/issues/1406

### Expected behaviour

- `ideviceinfo.exe -n -u <udid>`: must return the device's info.
- `idevicebackup2.exe -n -u <udid> backup <path>`: must start the backup.

mexmer, 2023-03-30:
did you pair device with computer, after you switched to wifi sync?
you can try to unpair and pair device again.

didix21, 2023-03-30:
yes, I've paired and I'm having the same error.

didix21, 2023-03-31:
Enabling debug mode amb getting the following logs:

```
16:03:38 idevice.c:537 idevice_connect(): Unsupported address family 0x00
16:03:38 lockdown.c:623 lockdownd_client_new(): could not connect to lockdownd (device 00008020-001E04523EF1402E)
16:03:38 lockdown.c:659 lockdownd_client_new_with_handshake(): failed to create lockdownd client.
ERROR: Could not connect to lockdownd: Mux error (-8)
````

mexmer, 2023-04-01:
i think i have seen issue with similar description to your error, it's something to do with usbmux message formating, should be either under usbmuxd or libusbmux ... 

it might be related to this
https://github.com/libimobiledevice/libimobiledevice/issues/1248

didix21, 2023-04-20:
Seems the patch provided #1248 works thanks