libimobiledevice/libimobiledevice #1431: idevicebackup2 could not perform backup protocol version exchange - where is 1.3.1 source?

altjira, 2023-04-01:
I've found multiple closed issues with this problem #1240, #1112 , #1111, #1032, #1028, #1013, #1001, #1115:
idevicebackup2 -- could not perform backup protocol version exchange
All recommend compiling libimobiledevice latest source - 1.3.1. However, github is listing latest source as 1.3.0 -- there is no latest source for me to compile!

Fedora 37 using packages from repository
- libimobiledevice-1.3.0-7.fc37.x86_64
- libimobiledevice-utils-1.3.0-7.fc37.x86_64
- libusbmuxd-2.0.2-7.fc37.x86_64
- usbmuxd-1.1.1-8.fc37.x86_64
- ifuse-1.1.4-6.fc37.x86_64

iPhone 13 IOS 16.3

mexmer, 2023-04-01:
latest source is not latest tag
current source version is 1.3.1, it has not tag yet, because it has no official relese. so in fact you need to compile from current master branch.

LLuanZZ, 2023-04-06:
> fonte mais recente não é tag mais recente a versão atual do source é 1.3.1, ainda não tem tag, pois não tem lançamento oficial. então, na verdade, você precisa compilar a partir do branch master atual.

How do I do that?

mexmer, 2023-04-06:
there is build script from nikias for ubuntu, since you running fedora, you will need to change appropriate parts

https://www.dropbox.com/s/rc1byi27n38j655/limd-build-linux.sh

if you never built software from source, i would not recommend starting with this project.

LLuanZZ, 2023-04-06:
> existe um script de construção do nikias para o ubuntu, já que você está executando o fedora, você precisará alterar as partes apropriadas
> 
> [https://www.dropbox.com/s/rc1byi27n38j655/limd-build-linux.sh ](https://www.dropbox.com/s/rc1byi27n38j655/limd-build-linux.sh)
> 
> se você nunca criou um software a partir do código-fonte, não recomendo começar com este projeto.

In my case I use Ubuntu. So I have to put that code in the terminal?

mexmer, 2023-04-06:
ah, didn't notice you different person than op (he is using fedora), just download that script, and run it from terminal.

LLuanZZ, 2023-04-06:
> ah, não notei que você é uma pessoa diferente de op (ele está usando o fedora), basta baixar esse script e executá-lo no terminal.

it worked, I just have to thank you

altjira, 2023-04-08:
I cloned from Git, compiled, and installed, in order:
- libplist   
- libimobiledevice-glue
- libusbmuxd
- libimobiledevice
- usbmuxd
- ifuse

I had to set ldconfig and $PKG_CONFIG_PATH to get this to work.

Libimobiledevice gave a LOT of "deprecated" warnings. But I'm now having trouble getting my iPhone recognized when I plug it in. lsusb sees the device, but usb-devices does not show a driver. The udev rule is in place. There is no /var/run/usbmuxd. When I find the time I'll do more troubleshooting, but this is clearly an entirely different problem and this issue can be closed.

mexmer, 2023-04-09:
did you run usbmuxd? also don't forget don't uninstall ubsmuxd and libimobile packages from your distribution, if you running binaries compiled from source

altjira, 2023-04-09:
Yes, I removed all the packages. It took a lot of unused dependencies with it.
```
[ ~]$ sudo dnf erase libimobiledevice usbmuxd ifuse
Dependencies resolved.
====================================================================================================================================
 Package                                Architecture          Version                                 Repository               Size
====================================================================================================================================
Removing:
 ifuse                                  x86_64                1.1.4-6.fc37                            @fedora                  65 k
 libimobiledevice                       x86_64                1.3.0-7.fc37                            @fedora                 191 k
 usbmuxd                                x86_64                1.1.1-8.fc37                            @fedora                 149 k
Removing dependent packages:
 libimobiledevice-utils                 x86_64                1.3.0-7.fc37                            @fedora                 459 k
Removing unused dependencies:
 libplist                               x86_64                2.2.0-8.fc37                            @fedora                 184 k
 libusbmuxd                             x86_64                2.0.2-7.fc37                            @fedora                  79 k

Transaction Summary
====================================================================================================================================
Remove  93 Packages
```
Showing select usb info:
```
[ ~]$ lsusb
Bus 001 Device 018: ID 05ac:12a8 Apple, Inc. iPhone 5/5C/5S/6/SE/7/8/X
[ ~]$ usb-devices

T:  Bus=01 Lev=01 Prnt=08 Port=03 Cnt=01 Dev#= 18 Spd=480  MxCh= 0
D:  Ver= 2.00 Cls=00(>ifc ) Sub=00 Prot=00 MxPS=64 #Cfgs=  4
P:  Vendor=05ac ProdID=12a8 Rev=14.05
S:  Manufacturer=Apple Inc.
S:  Product=iPhone
S:  SerialNumber=000081100010292E0268401E
C:  #Ifs= 0 Cfg#= 0 Atr= MxPwr=
cat: '/sys/bus/usb/devices/usb1/1-4/1-*:?.*/bInterfaceNumber': No such file or directory
cat: '/sys/bus/usb/devices/usb1/1-4/1-*:?.*/bAlternateSetting': No such file or directory
cat: '/sys/bus/usb/devices/usb1/1-4/1-*:?.*/bNumEndpoints': No such file or directory
cat: '/sys/bus/usb/devices/usb1/1-4/1-*:?.*/bInterfaceClass': No such file or directory
cat: '/sys/bus/usb/devices/usb1/1-4/1-*:?.*/bInterfaceSubClass': No such file or directory
cat: '/sys/bus/usb/devices/usb1/1-4/1-*:?.*/bInterfaceProtocol': No such file or directory
/usr/bin/usb-devices: line 91: printf: 0x: invalid hex number
/usr/bin/usb-devices: line 91: printf: 0x: invalid hex number
I:  If#= 0 Alt= 0 #EPs= 0 Cls=(none)() Sub= Prot= Driver=
```
usbmuxd doesn't run as user, but with sudo get the following:
```
[ ~]$ usbmuxd -f -v
[07:51:50.585][3] usbmuxd v1.1.1-46-gd0cda19 starting up
[07:51:50.585][0] Could not open lockfile
[ ~]$ sudo usbmuxd -f -v
[07:52:01.348][3] usbmuxd v1.1.1-46-gd0cda19 starting up
[07:52:01.348][4] Creating socket
[07:52:01.348][4] Listening on /var/run/usbmuxd
[07:52:01.348][4] Initializing USB
[07:52:01.348][3] Using libusb 1.0.25
[07:52:01.358][4] Registering for libusb hotplug events
[07:52:01.358][4] Found new device with v/p 05ac:12a8 at 1-18
[07:52:01.358][4] Requesting current mode from device 1-18
[07:52:01.358][4] 1 device detected
[07:52:01.358][3] Initialization complete
[07:52:01.361][4] Received response 3:3:3:0 for get_mode request for device 1-18
[07:52:01.361][2] Skipping switch device 1-18 mode from 1 to 1
libusb: warning [op_get_configuration] device unconfigured
[07:52:01.361][3] Found usbmux interface for device 1-18: 1
[07:52:01.361][4] Found interface 1 with endpoints 04/85 for device 1-18
[07:52:01.361][3] Changing configuration of device 1-18: 0 -> 4
[07:52:01.418][4] Using wMaxPacketSize=512 for device 1-18
[07:52:01.418][4] USB Speed is 480 MBit/s for device 1-18
[07:52:01.421][4] Got lang ID 1033 for device 1-18
[07:52:01.426][4] Got serial '000081100010292E0268401E' for device 1-18
[... 5 lines cut ...]
[07:52:01.428][4] preflight_worker_handle_device_add: Device 00008110-0010292E0268401E is not paired with this host.
[07:52:01.429][4] preflight_worker_handle_device_add: Found iPhone 16.3 device 00008110-0010292E0268401E
[07:52:01.737][4] Client 13 accepted: usbmuxd[192487]
[07:52:01.738][4] Client 13 connection closed
[07:52:01.738][4] Client 13 is going to be disconnected: usbmuxd[192487]
[07:52:01.793][4] Client 13 accepted: usbmuxd[192487]
[07:52:01.793][4] Client connected to device 1 (2->50899)
[07:52:01.793][4] Client 12 is going to be disconnected: usbmuxd[192487]
[07:52:01.793][4] preflight_worker_handle_device_add: Waiting for user to trust this computer on device 00008110-0010292E0268401E
[07:52:04.193][1] RX transfer stalled for device 1-18
[07:52:04.193][3] Removed device 1 on location 0x10012
[07:52:04.193][4] Client 13 is going to be disconnected: usbmuxd[192487]
[07:52:04.794][4] preflight_worker_handle_device_add: Finished waiting for notification from device 00008110-0010292E0268401E, is_device_connected 0
```

then it hangs until I ^C
```
^C[08:04:45.535][3] Caught signal 2, exiting
[08:04:45.535][4] Event processing interrupted
[08:04:45.535][3] usbmuxd shutting down
[08:04:45.635][3] Shutdown complete
[ ~]$ 
```

Sometimes when I plug the iPhone in, it will show up in nemo as a photo device, and sometimes, especially if I have rebooted the machine and the phone, it will show up briefly as a file storage device. I was wrong about /var/run/usbmuxd not showing up - I was looking for a directory, and it's a file - which gets refreshed every time I run usbmuxd.

mexmer, 2023-04-09:
usbmuxd can be either run manually as root, or you can run it from systemd or other service manager.

when did you run usbmuxd -f it means, running on foreground not background. and it will need to run all time, otherwise connected device will not authorize and be properly recognized.

altjira, 2023-04-09:
Well, launching it through a udev rule didn't seem to do anything, and running it in the foreground didn't either.. That's kind of my problem with it. But an entirely different problem than the "backup protocol version exchange - error 1" that I had with distro packages, so I guess I need to post this is the very quiet usbmuxd issues.

mexmer, 2023-04-10:
honestly no idea what can be wrong with your configuration
on debian i removed usbmuxd, libimobile and related packages, then compiled all from source, and run usbmuxd in background from commandline, and it's working.

usbmuxd does communication with device, when it's running on foreground, you can see the log, and check what's going on ... there is no other reason, to run it on foreground.

that error from lsusb or rather usb-devices tho' seems like there is something wrong with your udev, that is not creating endpoints in usbfs

Merovius, 2023-07-11:
Would it be possible to cut another actual release? I'd expect Fedora to package a newer version, once it is actually released. The compilation process seems pretty involved, especially if you have to build and install multiple 3rd party libraries. It seems honestly kind of strange to me to expect people to build this from source, when a release would make things trivial for most people and stop this issue being repeatedly reported.

daaku, 2024-03-02:
A new release would be wonderful since it "just works" with the current iOS and hardware.