libimobiledevice/libimobiledevice #1433: Android build with ndk-pkg

aciereszko, 2023-04-15:
Hi,

I wanted to get an Android build, with latest dev version, using ndk-pkg

I saw @leleiu008 mention in another post that should be possible. I modified the ndk-pkg formula, to not get the 1.3.0. I basically removed the two lines pkg_set src.url and pkg_set src.sha - which I understood would tell ndk-pkg use the head of default branch - master. I could be wrong here.

So then when running ndk-pkg install with that formula:
`aciereszko@ubuntu-vm:~/Documents/libimobiledevice$ ndk-pkg install libimobiledevice --min-sdk-api-level=24`
(I also was getting an API error, so I had to pass in the upped min-sdk argument)

I get an error:

`=>> STEP 12 : list files of source top directory 
==> ls -l /home/aciereszko/.ndk-pkg/installing.d/android/24/libimobiledevice/src
total 248
drwxrwxr-x 4 aciereszko aciereszko   4096 Apr 15 18:10 3rd_party
-rw-rw-r-- 1 aciereszko aciereszko    294 Apr 15 18:10 AUTHORS
-rwxrwxr-x 1 aciereszko aciereszko    365 Apr 15 18:10 autogen.sh
drwxrwxr-x 2 aciereszko aciereszko   4096 Apr 15 18:10 common
-rw-rw-r-- 1 aciereszko aciereszko  11080 Apr 15 18:10 configure.ac
-rw-rw-r-- 1 aciereszko aciereszko  17988 Apr 15 18:10 COPYING
-rw-rw-r-- 1 aciereszko aciereszko  26434 Apr 15 18:10 COPYING.LESSER
drwxrwxr-x 2 aciereszko aciereszko   4096 Apr 15 18:10 cython
drwxrwxr-x 3 aciereszko aciereszko   4096 Apr 15 18:10 docs
-rw-rw-r-- 1 aciereszko aciereszko 107126 Apr 15 18:10 doxygen.cfg.in
-rwxrwxr-x 1 aciereszko aciereszko    545 Apr 15 18:10 git-version-gen
drwxrwxr-x 3 aciereszko aciereszko   4096 Apr 15 18:10 include
drwxrwxr-x 2 aciereszko aciereszko   4096 Apr 15 18:10 m4
-rw-rw-r-- 1 aciereszko aciereszko    476 Apr 15 18:10 Makefile.am
-rw-rw-r-- 1 aciereszko aciereszko  23913 Apr 15 18:10 NEWS
-rw-rw-r-- 1 aciereszko aciereszko   7893 Apr 15 18:10 README.md
drwxrwxr-x 2 aciereszko aciereszko   4096 Apr 15 18:10 src
drwxrwxr-x 2 aciereszko aciereszko   4096 Apr 15 18:10 tools

=>> STEP 13 : copy or fetch licenses 

=>> STEP 14 : change to build script directory 
==> cd /home/aciereszko/.ndk-pkg/installing.d/android/24/libimobiledevice/src

=>> STEP 15 : build for native 
🔔  no build0  function defined in formula, skipped.

=>> STEP 16 : do some preparation works 
==> sed -i 's/@echo/echo/g' common/Makefile.in
sed: can't read common/Makefile.in: No such file or directory
💔  package [libimobiledevice] install failed! you can go to directory[/home/aciereszko/.ndk-pkg/installing.d/android/24/libimobiledevice] to look at the details.
`
Any idea what I am royally screwing up :) ?

mexmer, 2023-04-16:
source uses autotools to generate from AM templates, then you run configure, that will create makefile and other files needed for build. so you need to use autogen and automake as described in instructions.

aciereszko, 2023-04-17:
So it's not possible to modify the ndk-pkg formula to have ndk-pkg do all that for me?

mexmer, 2023-04-17:
tbh. i never built library for android, but ndk-pkg is not part of original source, so you should ask whoever created that package
so most likely you will need to ask here

https://github.com/leleliu008/ndk-pkg

you might also want to check this issue from past https://github.com/libimobiledevice/libimobiledevice/issues/1061

emilyrobert34, 2026-08-22:
The Android NDK setup can be quite useful for projects that need native code and better performance. I also find it helpful to keep supporting resources organized when working with technical or development-related tasks. For anyone researching public information [Spokane County Court](https://spokanecountycourts.org) can be a useful reference. Clear and reliable resources make it easier to find relevant details and keep research organized.