libimobiledevice/libimobiledevice #1435: Will there be a new release?

p-linnane, 2023-04-21:
Hello 👋 . I'm a maintainer for the [Homebrew](https://brew.sh) project. While packaging the new release of libplist we ran into some failures with libimobiledevice during regression testing. We attempted to patch libimobiledevicec with https://github.com/libimobiledevice/libimobiledevice/commit/eda2c5ea71829f11d69342e6858d09aa53943938, but it doesn't apply cleanly. I wanted to ask if there will be a new release any time soon. Thank you.

Relates to https://github.com/Homebrew/homebrew-core/pull/128974

nikias, 2023-04-21:
Yes, working on a release. Just wanted to see the build test succeeding before.

p-linnane, 2023-04-22:
Thank you! I'll keep an eye out for it.

p-linnane, 2023-05-02:
@nikias Hi. Just following up on the status of the new release. Thanks!

p-linnane, 2023-05-15:
@nikias Sorry to ping again. We are having a cascading effect of being unable to package libplist, libirecovery, and libimobiledevice-glue until this new release it cut. Is there a timeline?

chenrui333, 2023-05-26:
gentle ping @nikias :)

closed 2023-06-20 by p-linnane

jpstotz, 2023-06-20:
@p-linnane Could you please explain why you have closed this issue?

p-linnane, 2023-06-20:
> @p-linnane Could you please explain why you have closed this issue?

Lack of response for 2 months despite multiple follow ups.

jpstotz, 2023-06-20:
@p-linnane You are really giving up early ;) 

May be asking for a release was the wrong approach. If I understand it correctly a git tag in each repo would be sufficient for brew packaging. So may be we should better ask for "periodic" tagging (may be every few months)? 
Tagging should be scriptable which means very little effort for the libimobiledevice authors. 

Even if not every tag ends up in a brew release, that would still be better than the current situation.

FelixSchwarz, 2023-06-21:
> I am fixing a few things and will push a new tag. Not a fan of automatic tagging, also Linux distributions need an official release package.

At least in Fedora it is not too uncommon to package also from tags or even just git commits even though a proper release is of course much more welcome. Often in Fedora we package git snapshots when upstream has important commits but did not do a release for a long time. (Only mentioning Fedora here because this is the distro I know best.)

I can only encourage you to release as often as you can justify from the work load so users can benefit from bug fixes you did :-) 
Also you might get more meaningful bug reports from distributions if the released version is closer the master branch - otherwise half of the problems might be fixed already in master (but take time to diagnose) and few users might be able to compile their own version to reproduce on master.

chenrui333, 2023-07-04:
> I am fixing a few things and will push a new tag. Not a fan of automatic tagging, also Linux distributions need an official release package.

semantic releases are mostly kind of contract thing (indeed a bit ritual stuff), give us a ping once the new release is out. and thanks for the work!!

chenrui333, 2023-07-17:
Another two weeks passed :)