libimobiledevice/libimobiledevice #1439: idevicebackup2 not working for my iPhone

kavinduwije, 2023-05-03:
iOS version : 16.3.1
libimobiledevice version : 1.3.0

Started "com.apple.mobilebackup2" service on port 62101.
Could not perform backup protocol version exchange, error code -1

Please kindly support!!!

mexmer, 2023-05-04:
you are using outdated version of libimobile. you need to compile it from source code.

gpw928, 2023-07-31:
I have iOS 16.5.1(c).  I can't make the libimobiledevice version : 1.3.0 packages for Debian (11 and 12), Ubuntu 22.04.2, or  FreeBSD (12 or 13) work.  The Linux variants all do a backup, but unback fails.  The FreeBSD version can't even see the iPhone when it's present on the USB bus.

What's more, the advice to compile from source is of no use.  On Debian 11 and 12, I can't even configure:
```
configure: error: Package requirements (libplist-2.0 >= 2.3.0) were not met:

Package dependency requirement 'libplist-2.0 >= 2.3.0' could not be satisfied.
Package 'libplist-2.0' has version '2.2.0', required version is '>= 2.3.0'
```
Plus, it looks like the libimobiledevice-glue-dev package is no longer available, and needs to be compiled from source.

I appreciate the effort that goes into this, but maybe the documentation needs to be updated to indicate status, and  actions required to get it working...

mexmer, 2023-08-02:
@gpw928 you need to compile libimobile from github source, packages available in debian are made from last official release tag, which does not have support for ios 14 and newer

gpw928, 2023-08-03:
I cloned all the repositories mentioned on https://github.com/libimobiledevice, and set up the environment as follows:
```
sudo mkdir -p /opt/local
PKG_CONFIG_PATH=/opt/local/lib/pkgconfig/
export PKG_CONFIG_PATH
```
I configured them all with "./autogen.sh --prefix=/opt/local" (reverse order to that shown on the web page), made, and installed everything into the /opt/local tree.

I backed up the iPhone without issue (just as I had with the Debian 10 and 11 packages):
```
mkdir -p iPhone
idevicebackup2 backup --full  iPhone
```
and 
`idevicebackup2 unback  iPhone`
failed at exactly the same spot as Debian 10 and 11 packages:
```
Copying '00008101-001634823CE9003A/35/350ae8c3d242a0184fef82320c04c6e850f3c55a' to '_unback_/00008101-001634823CE9003A/var/root/Library/Preferences/wifid.plist'
ERROR: Could not receive from mobilebackup2 (-256)
Unback Aborted.
```
I don't know if it matters, but I subsequently found the usbmuxd dead.  I restarted it, and retried the "unback".  Same result and the usbmuxd was running throughout.

I'm happy to provide any additional information if that would help...

mexmer, 2023-08-03:
unback was broken by apple in ios version 10, it's phone os function, not libimobile function. and no, it will not work until apple fix it themself (which you can guess, almost 7 years, most likely not gona happen)

gpw928, 2023-08-03:
Thank you.  Is there any way, other than restoring the files to an iPhone, to access the files (photos, contacts) held in the backup? 
I don't have access to any Windows hosts.

mexmer, 2023-08-03:
there are 3rd party tools for browsing backups, i have not tested any tool for linux, but this seems promising

https://github.com/MaxiHuHe04/iTunes-Backup-Explorer

you could probly find many others, search for "itunes backup browser linux", you can probly also try with keyword viewer or extractor instead of browser

gpw928, 2023-08-04:
Thanks for that.  I'll see what I can get working.