libimobiledevice/libimobiledevice #1441: ERROR: Not able to cross-compile libimobiledevice on ubuntu

Shyamal-Shah, 2023-05-05:
I Ran `autogen.sh` and then `make` but make is giving openssl library not found error.

`./autogen.sh --host=x86_64-w64-mingw32 --without-cython openssl_LIBS="-L/bin -lssl -lcrypto"  openssl_CFLAGS=-I/include/openssl`

It ran successfully:
`Configuration for libimobiledevice 1.3.0-179-g860ffb7:
-------------------------------------------

  Install prefix: .........: /usr/local
  Debug code ..............: no
  Python bindings .........: no
  SSL support backend .....: OpenSSL

  Now type 'make' to build libimobiledevice 1.3.0-179-g860ffb7,
  and then 'make install' for installation.`
  
But when I ran make I am getting the following error:
`Making all in libsrp6a-sha512
make[3]: Entering directory '/usr/src/libimobiledevice/3rd_party/libsrp6a-sha512'
  CC       t_conv.lo
  CC       t_math.lo
t_math.c:36:11: fatal error: openssl/opensslv.h: No such file or directory
   36 | # include "openssl/opensslv.h"
      |           ^~~~~~~~~~~~~~~~~~~~
compilation terminated.
make[3]: *** [Makefile:432: t_math.lo] Error 1
make[3]: Leaving directory '/usr/src/libimobiledevice/3rd_party/libsrp6a-sha512'
make[2]: *** [Makefile:411: all-recursive] Error 1
make[2]: Leaving directory '/usr/src/libimobiledevice/3rd_party'
make[1]: *** [Makefile:465: all-recursive] Error 1
make[1]: Leaving directory '/usr/src/libimobiledevice'
make: *** [Makefile:395: all] Error 2`

What I have tried:
- I have installed libssl-dev
- I have tried reinstalling it
- I tried passing library location to the make command directly: `make -I/usr/include/openssl`

  Nothing has worked

P5-2005, 2023-05-25:
you are compiling for windows build right?, from linux host?
i tried just now on windows using msys2 mingw64, compiled no issue

tuchaVshortah, 2024-06-07:
> you are compiling for windows build right?, from linux host? i tried just now on windows using msys2 mingw64, compiled no issue

Hello. Do you still have instructions for msys2 and mingw64? I tried to build but get errors.