libimobiledevice/libimobiledevice #1446: Possible to interact with USB connected iPhone audio output?

dassoop, 2023-05-12:
I'm looking for a solution to route MacOS audio through USB to an iPhone's speaker output with no latency. Is this a possible use case for libimobile device? If not, do you have any clues?

nikias, 2023-05-12:
That wouldn't work. There is USB audio FROM the device, but not the other way.

dassoop, 2023-05-12:
What about using the lightening port as a microphone input? There are interfaces like the iRig Stream that use the lightening port as an audio input.

karl-police, 2024-07-15:
If it's possible to connect a idk to a phone https://www.youtube.com/watch?v=IgCcB1QgpYU

would it not be possible to temporarily use the default mic set on Windows?? :) :) :) :): ): ):)  ? ?? ?  ?????