libimobiledevice/libimobiledevice #1448: Could not perform backup protocol version exchange, error code -1

Corodin, 2023-05-15:
When I try to backup my device, I get this error message. No further explanation is given. I'm ready to give any information you may need.

mexmer, 2023-05-15:
you are using old version of libimobile, if you use search function, you will quickly find, there is multiple issues with same problem posted here.

you need to compile current version from source.

RFDAJE, 2023-07-02:
I'm also having the same issue for a month with the latest git version. ifuse mount works, but `idevicebackup2 backup` failed with `Could not perform backup protocol version exchange, error code -1`.

what I've installed on my archlinux.
- local/ifuse-git 2:1.1.4.r5.g814a0e3-2
- local/libimobiledevice-git 1:1.3.0.r186.g52ab7b7-1
- local/libimobiledevice-glue-git r46.214bafd-1
- local/libirecovery-git 2:1.1.0.r1.gc7b488f-1
- local/libplist-git 1:2.3.0.r18.g2d8d7ef-1
- local/libusbmuxd-git 2.0.2.r31.g8d30a55-1
- local/usbmuxd-git 1:1.1.1.r48.g01c94c7-1

mexmer, 2023-07-03:
@RFDAJE you need to compile from source on github, not distribution files

RFDAJE, 2023-07-03:
> @RFDAJE you need to compile from source on github, not distribution files

they are not binary from the distribution. the *-git in the package name means all compiled from git source, the gXXXXXX in the name means the git version being built against.

mexmer, 2023-07-03:
i see, so it's shortened on the end.
then check if somewhere on your path doesn't exists another idevicebackup2 binary, because i just tested current version on debian without any issues.

another thing you can try is unpair your device, and pair again (just in case escrowbag in lockdown file is somehow outdated).

RFDAJE, 2023-07-03:
> check if somewhere on your path doesn't exists another idevicebackup2 binary
I have idevicebackup and idevicebackup2 in PATH, what is the difference between them?

mexmer, 2023-07-03:
idevicebackup is for old ios, idevicebackup2 is for ios 5 and newer.

nikias, 2023-07-03:
> idevicebackup is for old ios, idevicebackup2 is for ios 5 and newer.

iOS 4 and newer :)

mexmer, 2023-07-04:
> > idevicebackup is for old ios, idevicebackup2 is for ios 5 and newer.
> 
> iOS 4 and newer :)

it's long time ago, didn't exactly remember when apple changed protocol.

RFDAJE, 2023-07-05:
What I have tried so far, issue persists.

```
# idevicepair unpair
SUCCESS: Unpaired with device 00000000-000XXXXXXXXXXXX

# idevicepair pair
SUCCESS: Paired with device 00000000-000XXXXXXXXXXXX

# which idevicebackup2
/usr/bin/idevicebackup2

# idevicebackup2 -v
idevicebackup2 1.3.0-186-g52ab7b7

# idevicebackup2 backup --full /mnt/backup/iPhoneXsMax/
Backup directory is "/mnt/backup/iPhoneXsMax/"
Started "com.apple.mobilebackup2" service on port 50324.
Negotiated Protocol Version 2.1
Reading Info.plist from backup.
Starting backup...
Enforcing full backup from device.
Backup will be unencrypted.
Requesting backup from device...
Incremental backup mode.
Sending '00000000-000XXXXXXXXXXXX/Status.plist' (189 Bytes)
Sending '00000000-000XXXXXXXXXXXX/Manifest.plist' (177.3 KB)
Sending '00000000-000XXXXXXXXXXXX/Manifest.db' (321.0 MB)
Sending '00000000-000XXXXXXXXXXXX/Status.plist' (189 Bytes)

### after several minutes

ERROR: Could not receive from mobilebackup2 (-4)
Received 0 files from device.
Backup Aborted.

# /usr/bin/idevicebackup2 backup --full /mnt/backup/iPhoneXsMax/
Backup directory is "/mnt/backup/iPhoneXsMax/"
Started "com.apple.mobilebackup2" service on port 50350.
Could not perform backup protocol version exchange, error code -1
```

mexmer, 2023-07-06:
320MB is a lot, do you have large number of photos inside phone? if that's so, you would probly need to modify libimobiledevice to increase actual timeout waiting for response. 

in our code (which was reverse engineered, but also is partially based on libimobile), we have 10 minute timeout for first device response after meta files), simply because we have customers with 512GB storage iphones, that have over 200 thousands pictures in them. and yes, it does take very long to backup service, to enumerate all photos.

other possibility is, that backup service has actually crashed (yes it happens), you can see that if you pull out syslog, right after unsuccesfull backup.
sometimes restarting device help, after service crash, but not always.

you need to be aware, that backup is controlled by device, and service running inside device, which says what computer needs to do. so if something breaks inside device (either because apple just did bad job, or because there is not enough resources), it's not a thing that libimobile can fix. still, thanks to logs, we might be able to figure out workaround.

RFDAJE, 2023-07-07:
> 320MB for manifest database is a lot, do you have large number of photos inside phone? if that's so, you would probly need to modify libimobiledevice to increase actual timeout waiting for response.
> 

In which file and on which line I can change the timeout value? what is the disadvantage or issue if the value had been given to a larger value?

mexmer, 2023-07-07:
should be here i think (not sure)

https://github.com/libimobiledevice/libimobiledevice/blob/6fc41f57fc607df9b07446ca45bdf754225c9bd9/src/property_list_service.c#L270
it's set to 30 second

well there is no inherent disadvantage, except if device stops responding, it will take longer, before you are notified about it.

still i would first check syslog, if by any chance service doesn't crash, in that case, this modification will not help.