libimobiledevice/libimobiledevice #1450: How to get jailbreak Status?

saliholoji, 2023-05-24:
Hello Guys,
Is there a way to get the Jailbreak status?
If not, please add. Thank you!

saliholoji, 2023-05-24:
I think I understand, if I type ideviceinfo, if Firmware Version is iBoot, there is no jailbreak, if Recovery Mode, there is jailbreak, right?

mexmer, 2023-05-24:
no, device being in recovery mode, doesn't signify jailbreak.

long story short, there is no universal method for detecting jailbreaks.

one thing you can try is to run selfsigned binary on device, if it runs, it's jailbroken (but then again, you will need to search for artifacts on device, to detect, which method was used for jailbreak)

there are also "virtual" jailbreaks, which don't allow running unsigned apps per se, but allow you to install applications from alternate store, those cannot be detected by method described above, because app actualy need to be signed by certificate, that is part of mdm profile used by particular jailbreak method.

some jailbreaks also only run/enable system services (like rooted afc, or some remote access code), but do not leave artifacts on filesystem (except some traces in logs)

so when you are testing if device is jailbroken, you need to search for mutliple stuff and check multiple things (including processes running in device, files on device filesystem, and device diagnostic logs).

P5-2005, 2023-05-25:
> Then in practice it is not possible to see it by looking with a single remote, thank you

there is workaround using ipsw from blacktop, if you can use afcclient with limbd, like here on my device 14.6 jailbroken before and cydia installed, but for now not jailbreak after restart 
```
C:\Users\P5_2005>ipsw idev afc ls ../../../../User
   ⨯ permission denied

C:\Users\P5_2005>ipsw idev afc ls ../../../../lib
   ⨯ permission denied

C:\Users\P5_2005>ipsw idev afc ls ../../../../
usr
boot
bin
._Library
.mount_rw
jbloader
sbin
.file
User
etc
System
var
Library
.bootstrapped
._bin
mnt
.fseventsd
private
._usr
lib
dev
.ba
.mb
tmp
Applications
Developer
cores

C:\Users\P5_2005>
```
when you see permission denied mean not jailbroken, but if you jailbreak then you can list /Applications
same if rootless just try to find the folders or files add by jailbreak first place

mexmer, 2023-05-25:
might be good idea, since on windows you can't use ifuse.
simple client for listing folders, and probly reading/writing single file.